sarxos/webcam-capture #713: macOS Mojave integration issues

msgilligan, 2019-05-03:
I'm having trouble getting camera input on macOS Mojave due to a requirement for [requesting permission](https://developer.apple.com/documentation/avfoundation/cameras_and_media_capture/requesting_authorization_for_media_capture_on_macos).

I'm not sure what combination of changes I need to make so that the OS knows to prompt the user for permission. It may require code-signing. If anyone has experience with this, please let me know. If not, I'll share information here when I figure it out.

If changes to `webcam-capture` are required we should make them. If not, we should at least document what needs to be done to make it work.

msgilligan, 2019-05-03:
So, it looks like all you have to do to get it working is add the following to your `Info.plist`:

	<key>NSCameraUsageDescription</key>
	<string>Uses camera to YOUR USE CASE HERE</string>

closed 2019-05-07 by msgilligan

algakzru, 2019-07-28:
@msgilligan 
I have put those lines into `/Applications/IntelliJ IDEA CE.app/Contents/Info.plist`, but asking for camera permission dialog does not pop up.