sarxos/webcam-capture #715: Maven compilation fails since https://swt-repo.googlecode.com/svn/repo/ no longer available

loquagg02, 2019-05-12:
`mvn clean install -P full `
 fails since https://swt-repo.googlecode.com/svn/repo/ no longer available

Needs to use http://maven-eclipse.github.io/maven

loquagg02, 2019-05-12:
Change required in webcam-capture-addons/webcam-capture-addon-swt/pom.xml

-      <id>swt-repo</id>
-      <url>https://swt-repo.googlecode.com/svn/repo/</url>
+      <id>maven-eclipse-repo</id>
+      <url>http://maven-eclipse.github.io/maven</url>

sarxos, 2019-05-13:
Hi @loquagg02,

Thank you for this information. Would you like to send pull request to address this or shall I change it by myself?

loquagg02, 2019-05-13:
Would you be able to change it yourself?

sarxos, 2019-05-21:
Ho @loquagg02, this is now addressed in https://github.com/sarxos/webcam-capture/commit/5874815ca3b66423fef0ad430812bb9b0a0506a0. Thank you!

commit 5874815c (2019-05-21): Change address to SWT repo, fixes #715
closed 2019-05-21 by sarxos