sarxos/webcam-capture #720: package com.github.sarxos.webcam does not exist

stvely, 2019-06-17:
I am using NetBeans 8.2 and am trying to run your example "TakePictureExample.java". and I'm getting the error:
run:
java.lang.ExceptionInInitializerError
Caused by: java.lang.RuntimeException: Uncompilable source code - package com.github.sarxos.webcam does not exist
at TakePictureExample.(TakePictureExample.java:8)

I was using webcam-capture-0.3.10.jar, slf4j-api-1.7.2.jar and bridj-0.6.2.jar but after reading your answer to a previous issue, I downloaded webcam-capture-0.3.12.jar, slf4j-api-1.7.2.jar and bridj-0.7.0.jar updated my class-path and even rebooted.
I have another program that I wrote that has been using webcam-capture-0.3.10.jar for some time without issue but I found out it has ceased to function so I tried your sample and found the error above.
I appreciate any help you can give.

sarxos, 2019-06-18:
Hi @stvely,

1. Download this ZIP file: https://github.com/sarxos/webcam-capture/releases/download/webcam-capture-parent-0.3.12/webcam-capture-0.3.12-dist.zip
2. There is a webcam-capture JAR inside, together with all required dependencies inside `libs` deirectory.
3. Remove corresponding JARs from your classpath and add new ones. Please make sure to use only these versions which are in ZIP.

Or if you are using Maven, then simply add this dependency into your POM:

```xml
<dependency>
  <groupId>com.github.sarxos</groupId>
  <artifactId>webcam-capture</artifactId>
  <version>0.3.12</version>
</dependency>
```

The error you described is only generated within IDE when JAR is missing in classpath or JAR does not contain given package. I'm pretty sure there is something wrong within your NetBeans project, but I cannot give you any details since I never really used NetBeans IDE (I'm on Eclipse most of the time). I always encourage everyone to use Maven instead of managing dependencies manually. This is much faster and will help you avoid such issues because Maven updates classpath for you. You may take a look at his tutorial from NetBeans if you are willing to give it a try:

https://platform.netbeans.org/tutorials/nbm-maven-quickstart.html

stvely, 2019-06-19:
Thanks for the reply. I followed you suggestions and replaced the old 3.10 jar and lib dependencies with those found in the 3.12 zip file. I carefully reviewed my classpath and searched for any remaining 3.10 files.
I was still getting the error so I followed the Maven tutorial to try the POM.XML dependency you suggested and am getting the error shown in the attached image.
Are your eclipse projects still functioning? I have an application that was working just fine for a year or two and now it is not even though I have done anything to it. Just curious if yours are still working?
[image]

sarxos, 2019-06-21:
Hi @stvely,

Yes, it's still working perfectly fine.

[image]

Can you show me your Maven pom.xml file? This error states that module cannot be installed because another module is missing. I suppose it's a problem in pom.xml you have there.

stvely, 2019-06-21:
Thanks for your help. I have egg on my face. I found the issue was unrelated to the com.github.sarxos.webcam package. I'm still not sure why Netbeans can't see it but My program is working again so all is well.

vvenkat1241, 2022-08-31:
> Thanks for your help. I have egg on my face. I found the issue was unrelated to the com.github.sarxos.webcam package. I'm still not sure why Netbeans can't see it but My program is working again so all is well.

please share on how you solved the issue?

stvely, 2022-08-31:
I am really sorry. This was 3 years ago and I can't remember what the fix
was. Apparently, it wasn't related to the sarxos.webcam package.

vvenkat1241, 2022-08-31:
which IDE were you using?

stvely, 2022-08-31:
Probably Netbeans but it was a project outside my normal scope.

vvenkat1241, 2022-08-31:
if the solution comes back to your memory please do share here
I am using eclipse though

stvely, 2022-08-31:
Will do.