sarxos/webcam-capture #721: set focus

gigirizzi, 2019-06-21:
Hi, i try to make a photo with java  but the focus is wrong, there is a method? how can make a fix? Tnx

sarxos, 2019-06-21:
Hi @gigirizzi 

No, there is no method to set focus. Camera's hardware (or driver) can do that automatically.

Things to try:

* If you have a dedicated software for your camera, turn it on and check if there is an option to enable auto focus.
* Clean your lens. If there is a glue or some other contaminants there the focus may be set incorrectly.
* Put more light on the object. Camera may have difficulties to set focus automatically if object is too dark.
* Google it, e.g. "enable logitech webcam autofocus".

If you are on Linux then you may use `v4l2-ctl` to list controls and set `focus_auto` and `exposure_auto` to respective values.

sarxos, 2019-06-21:
Here is a YouTube video on how to **disable** autofocus on Linux, but the same procedure can be applied to enable it again:

https://www.youtube.com/watch?v=7SZBQ5bqaWU