sarxos/webcam-capture #722: Intel crash on getImage()

brianmichalk1, 2019-06-25:
I develop on AMD64, but trying to run on Intel causes a crash:
`
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f3ca7c7fd24, pid=2668, tid=0x00007f3c5aa9e700
#
# JRE version: OpenJDK Runtime Environment (8.0_212-b03) (build 1.8.0_212-8u212-b03-0ubuntu1.16.04.1-b03)
# Java VM: OpenJDK 64-Bit Server VM (25.212-b03 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# V  [libjvm.so+0x793d24]
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread (0x00007f3c34003000):  JavaThread "atomic-processor-1" daemon [_thread_in_vm, id=2688, stack(0x00007f3c5a99e000,0x00007f3c5aa9f000)]

siginfo: si_signo: 11 (SIGSEGV), si_code: 2 (SEGV_ACCERR), si_addr: 0x00007f3c2c0f6000

Registers:
RAX=0x00007f3c2c170fe8, RBX=0x00007f3c34003000, RCX=0x00000000ec5ee7a8, RDX=0xffffffffffff0a04
RSP=0x00007f3c5aa9d528, RBP=0x00007f3c5aa9d570, RSI=0x0000000000000000, RDI=0x00007f3c2c08fff0
R8 =0x00007f3c5aa9d5a0, R9 =0x0000000000000010, R10=0x00007f3c911e67e7, R11=0x00007f3c911e6798
R12=0x00000000000e1000, R13=0x0000000000000000, R14=0x00007f3c5aa9d5a0, R15=0x0000000000000010
RIP=0x00007f3ca7c7fd24, EFLAGS=0x0000000000010282, CSGSFS=0x0000000000000033, ERR=0x0000000000000004
  TRAPNO=0x000000000000000e

Top of Stack: (sp=0x00007f3c5aa9d528)
0x00007f3c5aa9d528:   00007f3ca7f32c57 00007f3c5aa9d670
0x00007f3c5aa9d538:   00007f3c2c08fff0 0000000100028078
0x00007f3c5aa9d548:   00007f3ca4777ce8 0000000000000000
0x00007f3c5aa9d558:   00007f3c5aa9d5d8 00007f3c5aa9d6d0
0x00007f3c5aa9d568:   00007f3c34003000 00007f3c5aa9d5c0
0x00007f3c5aa9d578:   00007f3c911e6861 00000000000e1000
0x00007f3c5aa9d588:   00000000ed010178 0000000000000000
0x00007f3c5aa9d598:   00007f3ca7d34555 00000000ec50d7a0
0x00007f3c5aa9d5a8:   00007f3c34003000 0000000000000000
0x00007f3c5aa9d5b8:   00007f3ca4858410 00007f3c5aa9d660
0x00007f3c5aa9d5c8:   00007f3c91007ffd 00007f3c91007ffd
0x00007f3c5aa9d5d8:   00000000000e1000 00007f3ca4858468
0x00007f3c5aa9d5e8:   0000000000000010 00007f3ca4858410
0x00007f3c5aa9d5f8:   00000000ec50d7a0 00007f3c2c08fff0
0x00007f3c5aa9d608:   00007f3c910134f9 0000000000000000
0x00007f3c5aa9d618:   00000000ed010178 00007f3c5aa9d620
0x00007f3c5aa9d628:   00007f3ca485843a 00007f3c5aa9d6d0
0x00007f3c5aa9d638:   00007f3ca4859008 0000000000000000
0x00007f3c5aa9d648:   00007f3ca4858468 00007f3c5aa9d5d8
0x00007f3c5aa9d658:   00007f3c5aa9d690 00007f3c5aa9d718
0x00007f3c5aa9d668:   00007f3c91007ffd 00000000000e1000
0x00007f3c5aa9d678:   0000000000000000 0000000000000010
0x00007f3c5aa9d688:   0000000000000000 00000000000e1000
0x00007f3c5aa9d698:   00007f3ca7b4a063 0000000000000000
0x00007f3c5aa9d6a8:   00007f3c91007ffd 0000000000000010
0x00007f3c5aa9d6b8:   00007f3c91007b10 00000000ec50d7a0
0x00007f3c5aa9d6c8:   00007f3c2c08fff0 00000000ec2b1110
0x00007f3c5aa9d6d8:   00007f3c5aa9d6d8 00007f3ca48eed95
0x00007f3c5aa9d6e8:   00007f3c5aa9d758 00007f3ca48f2408
0x00007f3c5aa9d6f8:   0000000000000000 00007f3ca48eedc8
0x00007f3c5aa9d708:   00007f3c5aa9d690 00007f3c5aa9d740
0x00007f3c5aa9d718:   00007f3c5aa9d7a0 00007f3c91007d80 

Instructions: (pc=0x00007f3ca7c7fd24)
0x00007f3ca7c7fd04:   74 d1 08 48 83 c2 01 75 f0 c3 66 90 48 8b 74 d0
0x00007f3ca7c7fd14:   e8 48 89 74 d1 e8 48 8b 74 d0 f0 48 89 74 d1 f0
0x00007f3ca7c7fd24:   48 8b 74 d0 f8 48 89 74 d1 f8 48 8b 34 d0 48 89
[... 697 lines cut ...]
core id		: 1
cpu cores	: 2
apicid		: 3
initial apicid	: 3
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb ssbd ibrs ibpb stibp kaiser tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds
bogomips	: 4989.09
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:

Memory: 4k page, physical 3922528k(2289284k free), swap 4143100k(4143100k free)

vm_info: OpenJDK 64-Bit Server VM (25.212-b03) for linux-amd64 JRE (1.8.0_212-8u212-b03-0ubuntu1.16.04.1-b03), built on Apr 26 2019 02:02:50 by "buildd" with gcc 5.4.0 20160609

time: Tue Jun 25 18:01:04 2019
timezone: CDT
elapsed time: 4 seconds (0d 0h 0m 4s)

`

brianmichalk1, 2019-06-26:
More information:
The crash above occurs with a DFK23U16 from theimagingsource.com.  If I use a cheap ELP USB camera, this crash does not occur.

craigraw, 2020-12-20:
Getting reports that look like the same crash:

```
Thread 35 Crashed:: Java: atomic-processor-1
0   libsystem_kernel.dylib        	0x00007fff752862c6 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff75341bf1 pthread_kill + 284
2   libsystem_c.dylib             	0x00007fff751f06a6 abort + 127
3   libjvm.dylib                  	0x00000001010d3a64 os::abort(bool, void*, void const*) + 22
4   libjvm.dylib                  	0x00000001012839c0 VMError::report_and_die(int, char const*, char const*, __va_list_tag*, Thread*, unsigned char*, void*, void*, char const*, int, unsigned long) + 2836
5   libjvm.dylib                  	0x0000000101282e86 VMError::report_and_die(Thread*, unsigned int, unsigned char*, void*, void*, char const*, ...) + 152
6   libjvm.dylib                  	0x0000000101283a0d VMError::report_and_die(Thread*, unsigned int, unsigned char*, void*, void*) + 33
7   libjvm.dylib                  	0x00000001010d737a JVM_handle_bsd_signal + 804
8   libjvm.dylib                  	0x00000001010d4f4c signalHandler(int, __siginfo*, void*) + 45
9   libsystem_platform.dylib      	0x00007fff75336b5d _sigtramp + 29
10  libjvm.dylib                  	0x00000001011ee0eb SymbolTable::do_lookup(char const*, int, unsigned long) + 83
11  libjvm.dylib                  	0x000000010124d5cd Unsafe_CopyMemory0(JNIEnv_*, _jobject*, _jobject*, long, _jobject*, long, long) + 157
12  ???                           	0x00000001124bed94 0 + 4601933204
13  ???                           	0x000000010b0dca8c 0 + 4480420492
14  ???                           	0x000000010a9a7d22 0 + 4472864034
15  ???                           	0x000000010a9a7ee7 0 + 4472864487
16  ???                           	0x000000010a9a7f2c 0 + 4472864556
17  ???                           	0x000000010a9a7ee7 0 + 4472864487
18  ???                           	0x000000010a9a7f2c 0 + 4472864556
19  ???                           	0x000000010a99e949 0 + 4472826185
20  libjvm.dylib                  	0x0000000100e45f02 JavaCalls::call_helper(JavaValue*, methodHandle const&, JavaCallArguments*, Thread*) + 598
21  libjvm.dylib                  	0x0000000100e452ec JavaCalls::call_virtual(JavaValue*, Klass*, Symbol*, Symbol*, JavaCallArguments*, Thread*) + 254
22  libjvm.dylib                  	0x0000000100e453a8 JavaCalls::call_virtual(JavaValue*, Handle, Klass*, Symbol*, Symbol*, Thread*) + 98
23  libjvm.dylib                  	0x0000000100ec3090 thread_entry(JavaThread*, Thread*) + 120
24  libjvm.dylib                  	0x000000010122d590 JavaThread::thread_main_inner() + 134
25  libjvm.dylib                  	0x000000010122d3ce JavaThread::run() + 280
26  libjvm.dylib                  	0x000000010122b199 Thread::call_run() + 113
27  libjvm.dylib                  	0x00000001010d332e thread_native_entry(Thread*) + 297
28  libsystem_pthread.dylib       	0x00007fff7533f2eb _pthread_body + 126
29  libsystem_pthread.dylib       	0x00007fff75342249 _pthread_start + 66
30  libsystem_pthread.dylib       	0x00007fff7533e40d thread_start + 13
```