sarxos/webcam-capture #732: issue  java.lang.NoClassDefFoundError: org/slf4j/LoggerFactory

DurankGts, 2019-07-26:
Please help me.

this error
java.lang.NoClassDefFoundError: org/slf4j/LoggerFactory
	at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:101)
	at com.codename1.camerakit.impl.CameraNativeAccessImpl.getWebcam(Unknown Source)
	at com.codename1.camerakit.impl.CameraNativeAccessImpl.start(Unknown Source)
	at com.codename1.camerakit.CameraKit.start(CameraKit.java:72)
	at com.codename1.camerakit.demo.CameraKitDemo.start(CameraKitDemo.java:95)
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at com.codename1.impl.javase.Executor$3$1.run(Executor.java:256)
	at com.codename1.ui.Display.processSerialCalls(Display.java:1298)
	at com.codename1.ui.Display.mainEDTLoop(Display.java:1093)
	at com.codename1.ui.RunnableWrapper.run(RunnableWrapper.java:120)
	at com.codename1.impl.CodenameOneThread.run(CodenameOneThread.java:176)
Caused by: java.lang.ClassNotFoundException: org.slf4j.LoggerFactory
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 14 more

tirz, 2019-08-09:
This thread would be more appropriate on StackOverflow instead of GitHub.

You need to install slf4j.

You can find some help on the web: https://stackoverflow.com/questions/12926899/java-lang-noclassdeffounderror-org-slf4j-loggerfactory

sarxos, 2021-01-29:
See https://github.com/sarxos/webcam-capture/wiki/java.lang.ClassNotFoundException%3A-org.slf4j.LoggerFactory

closed 2021-01-29 by sarxos