sarxos/webcam-capture #733: plz help me

hemeshwari, 2019-08-03:
Exception in thread "main" java.lang.NoClassDefFoundError: org/slf4j/LoggerFactory
        at com.github.sarxos.webcam.Webcam.<clinit>(Webcam.java:97)
        at DetectWebcamExample.main(DetectWebcamExample.java:7)
Caused by: java.lang.ClassNotFoundException: org.slf4j.LoggerFactory
        at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(Unknown Source)
        at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(Unknown Source)
        at java.base/java.lang.ClassLoader.loadClass(Unknown Source)
        ... 2 more

tirz, 2019-08-09:
Duplicate of #732

sarxos, 2021-01-29:
See https://github.com/sarxos/webcam-capture/wiki/java.lang.ClassNotFoundException%3A-org.slf4j.LoggerFactory

closed 2021-01-29 by sarxos