sarxos/webcam-capture #742: Not Capturing Image on Windows 10

raphaelbn, 2019-10-04:
Hi,

I'm facing a problem with sarxos api. I'm using windows 10 and java 8, but the camera is black, like is not working, not capturing image.

When I use windows 8 or linux, the application works fine.

There is something to do?

My code is below:

`
public class App {
	
	public static void main(String[] args) throws InterruptedException {

		Webcam webcam = Webcam.getDefault();
		webcam.setViewSize(WebcamResolution.VGA.getSize());

		WebcamPanel panel = new WebcamPanel(webcam);
		panel.setFPSDisplayed(true);
		panel.setDisplayDebugInfo(true);
		panel.setImageSizeDisplayed(true);
		panel.setMirrored(true);

		JFrame window = new JFrame("Test webcam panel");
		window.add(panel);
		window.setResizable(true);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}
`

noorball, 2019-10-07:
This code works just fine for me (Windows 10, Java 8).
You can try to check permissions in Camera privacy settings in default Windows settings and in Vendor settings app (e.g. Lenovo Vantage etc).

raphaelbn, 2019-10-08:
It's so weird..
I am using desktop pc and I've already checked and setup the Camera privacy in Windows.
For me is showing a black panel.
Looks like the application can communicate with the camera, but could not show the image.
I don't know if it's a driver problem or hardware problem.
My camera is C922 Pro Stream Webcam.

[image]

raphaelbn, 2019-10-08:
In others application from Windows the image is showing.

raphaelbn, 2019-10-08:
The debug console:

`
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/C:/Users/raphael.nascimento/Downloads/logback-classic-1.0.9.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/C:/Users/raphael.nascimento/.m2/repository/org/slf4j/slf4j-log4j12/1.7.26/slf4j-log4j12-1.7.26.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [ch.qos.logback.classic.util.ContextSelectorStaticBinder]
09:46:07.748 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
09:46:07.755 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
09:46:08.275 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device C922 Pro Stream Webcam 0
09:46:08.275 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logi Capture 1
09:46:08.279 [main] DEBUG com.github.sarxos.webcam.Webcam - Setting new resolution 640x480
09:46:08.362 [main] DEBUG com.github.sarxos.webcam.WebcamPanel - Starting panel rendering and trying to open attached webcam
09:46:08.369 [SwingWorker-pool-2-thread-1] DEBUG com.github.sarxos.webcam.WebcamLock - Lock Webcam C922 Pro Stream Webcam 0
09:46:08.372 [atomic-processor-1] INFO  c.g.s.webcam.ds.cgt.WebcamOpenTask - Opening webcam C922 Pro Stream Webcam 0
09:46:08.372 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Opening webcam device C922 Pro Stream Webcam 0
09:46:08.373 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device 0 starting session, size java.awt.Dimension[width=640,height=480]
09:46:08.692 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device session started
09:46:08.694 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Clear memory buffer
09:46:11.280 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
09:46:35.854 [atomic-processor-1] DEBUG c.g.s.w.d.b.WebcamDefaultDevice - Webcam device com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice@7548be95 is now open
09:46:35.855 [SwingWorker-pool-2-thread-1] DEBUG com.github.sarxos.webcam.Webcam - Webcam is now open C922 Pro Stream Webcam 0
09:46:35.864 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device C922 Pro Stream Webcam 0
09:46:35.864 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logi Capture 1
09:46:38.864 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
09:46:41.326 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
09:46:41.332 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device C922 Pro Stream Webcam 0
09:46:41.333 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logi Capture 1
09:46:44.334 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
09:46:46.751 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
09:46:46.758 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device C922 Pro Stream Webcam 0
09:46:46.758 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logi Capture 1
09:46:49.759 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
09:46:52.168 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
09:46:52.174 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device C922 Pro Stream Webcam 0
09:46:52.175 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logi Capture 1
09:46:55.175 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
09:46:57.605 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
09:46:57.612 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device C922 Pro Stream Webcam 0
09:46:57.612 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logi Capture 1
09:47:00.613 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
09:47:03.028 [frames-refresher-[0]] ERROR c.g.s.w.d.b.WebcamDefaultDevice - Timeout when requesting image!
09:47:03.041 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device C922 Pro Stream Webcam 0
09:47:03.041 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Found device Logi Capture 1

`

raphaelbn, 2019-10-08:
I passed the parameter -client when run by command line e its works fine.
Like this:
"java -jar -client class.jar"

Why I need to pass this parameter?

raphaelbn, 2019-10-09:
Problem solved..
I was using Spring Tools Suite, its a IDE based on Eclipse, but the camera doesn't work in that.
Now I am using Eclipse IDE and the camera is working fine.

closed 2019-10-09 by raphaelbn