sarxos/webcam-capture #743: High CPU-Load with Linux after open

ifmec, 2019-10-05:
My topic is the CPU load

When I open() (synchronous/asynchronous) I have a high CPU load which is based only on the open. After close the load is gone . 
I use Java 7/8 and Linux 32 bit, the resolution of the camera is 640x480. The load is slightly lower in assynchronous mode than in synchronous mode.  I don't understand is that with my own Delaycalcolator with very high values the load doesn't decrease measurably. Why? The UseCase is: An image is created by request.  The image should be a maximum of 0.5 seconds old. 

However, an open/close int the request would be clearly too slow. 

Do you have any idea how I can reduce the load?
Many thanks in advance!

sebouh137, 2020-04-16:
The same seems to be true for MacOS.  My laptop starts heating up, and the fans start whirring out of control, and I'm afraid it might overheat if I leave it running too long.

sebouh137, 2020-04-17:
update:  I am using VisualVM to profile this, and it showing me that the following native method is the one that is taking up almost all the of CPU time:  
com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.nextFrame()

@sarxos , do you have any insight into this?