sarxos/webcam-capture #746: How to storing image in JPG, without java.lang.OutOfMemoryError ?

molina-henrique, 2019-11-01:
I'm looking for some solution to storing JPG from camera, 
I know that approach the code not is good! and facing memory leak in few seconds 

> java.lang.OutOfMemoryError: GC overhead limit exceeded  

```
public class WebcamPickerDemo {
	static {
		String rtsp = "rtsp://admin:password@address:port/";
		Webcam.setDriver(new VlcjDriver(Arrays.asList(new MediaListItem("Channel", rtsp, new ArrayList<MediaListItem>()))));
	}
	public static void main(String[] args) {
		Dimension size = WebcamResolution.QVGA.getSize();
		Webcam webcam = Webcam.getDefault();
		webcam.setViewSize(size);
		webcam.open(true);
		webcam.addWebcamListener(new WebcamListener() {
			public void webcamOpen(WebcamEvent we) { }
			public void webcamImageObtained(WebcamEvent we) {
				try {
// Propose is: save collected image to after another service save on mp4 like backup
					FileOutputStream fileout= new FileOutputStream( String.format("/Users/molines/20191101/%s.jpg",System.currentTimeMillis()) );
					ImageIO.write(we.getImage(), "JPG", fileout);
					fileout.flush();
					fileout.close();
				} catch (IOException e) {
					e.printStackTrace();
				}
			}
			public void webcamDisposed(WebcamEvent we) { }
			public void webcamClosed(WebcamEvent we) { }
		});
	}
}
```
Any suggestion ?
Thanks in advanced!

ysz0507, 2019-11-17:
Do you use a your computer or a raspberry pi?

molina-henrique, 2019-11-18:
computer