sarxos/webcam-capture #747: For some reason I can't access my webcam on the Raspberry Pi

ysz0507, 2019-11-10:
After a few problems with the api, which i solved successfully, now i really dont know how to fix this:
If i execute my code without the webcam plugged in everything is working as expected. But with my webcam plugged in i got the following exception:

My code:
```
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver;

public class DetectWebcamExample {
	static {
	     Webcam.setDriver(new V4l4jDriver()); // this is important
	}
	public static void main(String[] args) {
		Webcam webcam = Webcam.getDefault();
		if (webcam != null) {
			System.out.println("Webcam: " + webcam.getName());
		} else {
			System.out.println("No webcam detected");
		}
	}
}
```

The output with webcam:
```
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: java.lang.UnsatisfiedLinkError: au.edu.jcu.v4l4j.VideoDevice.doInit(Ljava/lang/String;)J
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:816)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:879)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:856)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:834)
	at DetectWebcamExample.main(DetectWebcamExample.java:14)
Caused by: java.util.concurrent.ExecutionException: java.lang.UnsatisfiedLinkError: au.edu.jcu.v4l4j.VideoDevice.doInit(Ljava/lang/String;)J
	at java.base/java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.base/java.util.concurrent.FutureTask.get(FutureTask.java:191)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 5 more
Caused by: java.lang.UnsatisfiedLinkError: au.edu.jcu.v4l4j.VideoDevice.doInit(Ljava/lang/String;)J
	at au.edu.jcu.v4l4j.VideoDevice.doInit(Native Method)
	at au.edu.jcu.v4l4j.VideoDevice.<init>(VideoDevice.java:271)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.<init>(V4l4jDevice.java:66)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.getDevices(V4l4jDriver.java:55)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.base/java.lang.Thread.run(Thread.java:844)

------------------
(program exited with code: 1)
Press return to continue
```
The Jars i used:
- slf4j-nop-1.6.1
- webcam-capture-driver-v4l4j-0.3.10-20140923.154112-11
- slf4j-api-1.7.2
- v4l4j-0.9.1-r507
- webcam-capture-0.3.10

My Hardware:
 - Logitech C 170 (Im pretty sure its compatible)
 - Raspberry Pi 3B with Raspbian 9 (stretch)

Thanks for helping

Grashalmbeisser, 2019-11-10:
I am having the same Problem, could you post yur answer when you find it out ? 
I am trying to solve this as well

ysz0507, 2019-11-10:
Of course no problem. Could u pls also post it if u find the solution before me?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Grashalmbeisser, 2019-11-18:
Are you also a student at uiniversity duisburg essen?  
If yes itr is a wrapper problem. The wrapper we got does not work with webcams ( except for the LegoCams)

yuripourre, 2019-11-18:
@ysz0507 Would you mind trying the [ffmpeg driver](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ffmpeg-cli)? It was reported to be working with Raspberry Pi.

ysz0507, 2019-11-20:
>Are you also a student at uiniversity duisburg essen?   ...

@Grashalmbeisser no sry i‘m not a student. In the moment i‘m visiting the 11th class. But thx for the advice.

ysz0507, 2019-11-20:
>@ysz0507 Would you mind trying the [ffmpeg driver](https://github.com/sarxos/webcam-capture/tree/master/webcam-capture-drivers/driver-ffmpeg-cli)? It was reported to be working with Raspberry Pi.

@yuripourre Sry I’ve just read this jet. Yes for sure, but i‘m not really confident because the same thing was written a few times about the V4L4J-Driver. But we will see maybe it will work. 👌🏻 thx for helping

yuripourre, 2019-11-21:
I belive the FFMpeg drive has more chances to work because it doesn't rely on JNI solutions, it may be less performant but since it uses Java Process it requires less configurations.

If you have issues, please, let me know.

ysz0507, 2019-11-21:
>I belive the FFMpeg drive has more chances to work because it doesn't rely on JNI solutions, it may be less performant but since it uses Java Process it requires less configurations. ...

@yuripourre 
Yes, it really works as i changed it! 😁😁 Thanks a lot! It was really helpful 😄

@Grashalmbeisser does it also work for your project?

closed 2019-12-18 by ysz0507