sarxos/webcam-capture #751: [Request] Publish ffmpeg driver to maven

yuripourre, 2019-11-21:
Hello @sarxos, I've been using the ffmpeg driver for two years now and had no issues so far. Some people reported that it works on Raspberry Pi.

Please, could you publish it to maven?

Thank you in advance

yuripourre, 2020-03-09:
As a temporary fix, I uploaded the driver under my domain:

```
<dependency>
    <groupId>com.harium</groupId>
    <artifactId>webcam-capture-driver-ffmpeg-cli</artifactId>
    <version>0.3.12</version>
</dependency>
```

sarxos, 2020-03-30:
Hi @yuripourre There are some fixes and tests I want to perform and if everything is find I think I will be releasing full set of 0.3.13 artifacts soon.

yuripourre, 2020-03-30:
Thank you @sarxos. Please tell me if I can help with anything.