sarxos/webcam-capture #752: Cant access webcam

pablobonetp, 2019-11-25:
Hey!

Im trying to use the webcam to capture QR. I just used this piece of code:

`	public static void main(String[] args) throws IOException 
	{

		Webcam webcam = Webcam.getDefault();
		webcam.open();
		BufferedImage image = webcam.getImage();
		ImageIO.write(image, "PNG", new File("1.png"));
	}`

It says it's opening the Camera on my Mac but then it doesn't work. I get this output:

`[main] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
[atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam FaceTime HD Camera CC25396MW1DG1HNB9
[frames-refresher-[CC25396MW1DG1HNB9]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!
[atomic-processor-1] WARN com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Null array pointer found instead of image
[atomic-processor-1] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Images bytes buffer is null!
Exception in thread "main" java.lang.IllegalArgumentException: image == null!
	at java.desktop/javax.imageio.ImageTypeSpecifier.createFromRenderedImage(ImageTypeSpecifier.java:925)
	at java.desktop/javax.imageio.ImageIO.getWriter(ImageIO.java:1608)
	at java.desktop/javax.imageio.ImageIO.write(ImageIO.java:1540)
	at App.main(App.java:16)
[shutdown-hook-1] INFO com.github.sarxos.webcam.WebcamShutdownHook - Automatic FaceTime HD Camera CC25396MW1DG1HNB9 deallocation
[shutdown-hook-1] INFO com.github.sarxos.webcam.Webcam - Disposing webcam FaceTime HD Camera CC25396MW1DG1HNB9
[frames-refresher-[CC25396MW1DG1HNB9]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!`

Nixonite, 2020-04-10:
I got the same issue!

I was trying this with Clojure and I also get nil when I try to use .getImage()

see my example:

`mygameidea.core=> (let [mycam (Webcam/getDefault)] (.open mycam) (javax.imageio.ImageIO/write (.getImage mycam) "PNG" (java.io.File. "yolo.png")))
Execution error (IllegalArgumentException) at javax.imageio.ImageTypeSpecifier/createFromRenderedImage (ImageTypeSpecifier.java:925).
image == null!`

Nixonite, 2020-04-10:
Update:

I added [org.slf4j/slf4j-simple "1.6.1"] to my dependencies in clojure and it produced a very different error now!

`mygameidea.core=> (.open mycam)
14937 [atomic-processor-1] INFO com.github.sarxos.webcam.ds.cgt.WebcamOpenTask - Opening webcam FaceTime HD Camera DJH9076XPLSGJKD1X
true
mygameidea.core=> (.getImage mycam)
49377 [frames-refresher-[DJH9076XPLSGJKD1X]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!
49378 [atomic-processor-1] WARN com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Null array pointer found instead of image
49378 [atomic-processor-1] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Images bytes buffer is null!
nil
mygameidea.core=> 57272 [frames-refresher-[DJH9076XPLSGJKD1X]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!
65265 [frames-refresher-[DJH9076XPLSGJKD1X]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!`

Nixonite, 2020-04-10:
@pablobonetp UPDATE!!!
I got it working at last. So there was a really funny issue actually. Two things.

First one - that might not be a problem at all, idk. I had graalvm running for clojure instead of the oracle JDK. I didn't test if graalvm works with this problem or not, but I uninstalled it and made sure that I was running Oracle JDK 1.8.

SECOND and MOST IMPORTANT. I was running my code via the repl built into VSCODE. When I ran outside of VSCODE on my Mac (Mojave) I realized that I suddenly got a popup asking for permission to view the webcam. I clicked OK and it worked like a charm! Not sure why it didn't show up from VSCODE's repl... damn that was a giant waste of time :( But it works yay! 

BTW I found the solution here http://deviatenow.com/ss/viewtopic.php?f=6&t=779&sid=d18dcdd7a55e543420d23c769bcb7e06&start=15