sarxos/webcam-capture #756: Getting Error in v4l4j driver on raspberrry  pi 4

lokendrayadav, 2019-12-31:
Hi ,
I am getting below error when using webcam-capture library with driver v4l4j on rpi 4 . in rpi 3 it is working file but in rpi 4 it is not working. please give any solution. 

[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[ v4l4j_ResolutionInfo.c:57 ] [V4L4J] Error looking up the palette with index 74
au.edu.jcu.v4l4j.exceptions.JNIException: Error looking up image format
	at au.edu.jcu.v4l4j.ResolutionInfo.doGetType(Native Method)
	at au.edu.jcu.v4l4j.ResolutionInfo.<init>(ResolutionInfo.java:145)
	at au.edu.jcu.v4l4j.ImageFormat.<init>(ImageFormat.java:84)
	at au.edu.jcu.v4l4j.ImageFormatList.listFormats(Native Method)
	at au.edu.jcu.v4l4j.ImageFormatList.<init>(ImageFormatList.java:111)
	at au.edu.jcu.v4l4j.DeviceInfo.getInfo(Native Method)
	at au.edu.jcu.v4l4j.DeviceInfo.<init>(DeviceInfo.java:254)
	at au.edu.jcu.v4l4j.VideoDevice.initDeviceInfo(VideoDevice.java:298)
	at au.edu.jcu.v4l4j.VideoDevice.<init>(VideoDevice.java:274)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.getVideoDevice(V4l4jDevice.java:105)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.<init>(V4l4jDevice.java:89)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.getDevices(V4l4jDriver.java:49)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
There was an error checking the supported resolutions.

satispande, 2020-01-16:
Hi @lokendrayadav  did you got any solution?

lokendrayadav, 2020-01-17:
> Hi @lokendrayadav did you got any solution?
Hi satishpande,
No I did not got any solution?

qdimka, 2020-02-20:
Hi,
I have the same issue, with v4l4j driver on rpi4.

Full log:

```
Feb 20, 2020 10:15:19 AM tornadofx.Stylesheet$Companion detectAndInstallUrlHandler
INFO: Installing CSS url handler, since it was not picked up automatically
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.github.sarxos.v4l4j.NativeUtils (file:/home/pi/Desktop/avastar-device/lib/avastar-device-0.0.jar) to field java.lang.ClassLoader.usr_paths
WARNING: Please consider reporting this to the maintainers of com.github.sarxos.v4l4j.NativeUtils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[ v4l4j_ResolutionInfo.c:57 ] [V4L4J] Error looking up the palette with index 74
au.edu.jcu.v4l4j.exceptions.JNIException: Error looking up image format
	at au.edu.jcu.v4l4j.ResolutionInfo.doGetType(Native Method)
	at au.edu.jcu.v4l4j.ResolutionInfo.<init>(ResolutionInfo.java:145)
	at au.edu.jcu.v4l4j.ImageFormat.<init>(ImageFormat.java:84)
	at au.edu.jcu.v4l4j.ImageFormatList.listFormats(Native Method)
	at au.edu.jcu.v4l4j.ImageFormatList.<init>(ImageFormatList.java:111)
	at au.edu.jcu.v4l4j.DeviceInfo.getInfo(Native Method)
	at au.edu.jcu.v4l4j.DeviceInfo.<init>(DeviceInfo.java:254)
	at au.edu.jcu.v4l4j.VideoDevice.initDeviceInfo(VideoDevice.java:298)
	at au.edu.jcu.v4l4j.VideoDevice.<init>(VideoDevice.java:274)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.getVideoDevice(V4l4jDevice.java:105)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.<init>(V4l4jDevice.java:89)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.getDevices(V4l4jDriver.java:49)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
There was an error checking the supported resolutions.
Please report this error to thev4l4j mailing list.
See REAME file for information on reporting bugs
[ v4l4j_ResolutionInfo.c:57 ] [V4L4J] Error looking up the palette with index 74
au.edu.jcu.v4l4j.exceptions.JNIException: Error looking up image format
	at au.edu.jcu.v4l4j.ResolutionInfo.doGetType(Native Method)
	at au.edu.jcu.v4l4j.ResolutionInfo.<init>(ResolutionInfo.java:145)
	at au.edu.jcu.v4l4j.ImageFormat.<init>(ImageFormat.java:84)
	at au.edu.jcu.v4l4j.ImageFormatList.listFormats(Native Method)
	at au.edu.jcu.v4l4j.ImageFormatList.<init>(ImageFormatList.java:111)
	at au.edu.jcu.v4l4j.DeviceInfo.getInfo(Native Method)
	at au.edu.jcu.v4l4j.DeviceInfo.<init>(DeviceInfo.java:254)
	at au.edu.jcu.v4l4j.VideoDevice.initDeviceInfo(VideoDevice.java:298)
	at au.edu.jcu.v4l4j.VideoDevice.<init>(VideoDevice.java:274)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.getVideoDevice(V4l4jDevice.java:105)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDevice.<init>(V4l4jDevice.java:89)
	at com.github.sarxos.webcam.ds.v4l4j.V4l4jDriver.getDevices(V4l4jDriver.java:49)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
There was an error checking the supported resolutions.
Please report this error to thev4l4j mailing list.
See REAME file for information on reporting bugs
[ v4l4j_ResolutionInfo.c:57 ] [V4L4J] Error looking up the palette with index 74
au.edu.jcu.v4l4j.exceptions.JNIException: Error looking up image format
	at au.edu.jcu.v4l4j.ResolutionInfo.doGetType(Native Method)
	at au.edu.jcu.v4l4j.ResolutionInfo.<init>(ResolutionInfo.java:145)
	at au.edu.jcu.v4l4j.ImageFormat.<init>(ImageFormat.java:84)
	at au.edu.jcu.v4l4j.ImageFormatList.listFormats(Native Method)
	at au.edu.jcu.v4l4j.ImageFormatList.<init>(ImageFormatList.java:111)
[... 69 lines cut ...]
Querying image format

============================================
Querying current image format settings

============================================
Querying supported frame sizes

============================================
Querying standard controls

Control: id: 0x98090c - name: White Balance, Automatic - min: 0 -max: 1 - step: 1 - type: 2(Boolean) - flags: 0 ()
Control: id: 0x980910 - name: Gamma - min: 500 -max: 3000 - step: 1 - type: 1(Integer) - flags: 32 (slider )
Control: id: 0x980914 - name: Horizontal Flip - min: 0 -max: 1 - step: 1 - type: 2(Boolean) - flags: 0 ()
Control: id: 0x980915 - name: Vertical Flip - min: 0 -max: 1 - step: 1 - type: 2(Boolean) - flags: 0 ()
============================================
Querying private controls

we shouldnt be here...
we shouldnt be here...
<....>
we shouldnt be here...
```

I think I should install additional native libraries. Am I right?

qdimka, 2020-02-20:
I found a problem.

On rpi4 we have /dev/video0 and /dev/video1 for one web camera. But we can't use /dev/video1 to capture video.

Webcam.getDefault() returns /dev/video1. 

I changed `Webcam.getDefault()` to `Webcam.getWebcamByName("/dev/video0")` and received another error.

```
Feb 20, 2020 1:55:03 PM tornadofx.Stylesheet$Companion detectAndInstallUrlHandler
INFO: Installing CSS url handler, since it was not picked up automatically
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.github.sarxos.v4l4j.NativeUtils (file:/home/pi/Desktop/avastar-device/lib/avastar-device-0.0.jar) to field java.lang.ClassLoader.usr_paths
WARNING: Please consider reporting this to the maintainers of com.github.sarxos.v4l4j.NativeUtils
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
[ libvideo.c:68 ] Using libvideo version 0.9-UNKNOWN
[ v4l2-input.c:67 ] The device /dev/video12 seems to be a valid V4L2 device but without capture capability.
[ v4l2-input.c:68 ] Please let the author know about this error.
[ v4l2-input.c:69 ] See the ISSUES section in the libvideo README file.
[ v4l2-input.c:70 ] Listing the reported capabilities:
Failed to get image format.: Invalid argument
```

qdimka, 2020-02-20:
@lokendrayadav I found solution. 

I checked `ls /dev/video*`

timmattison, 2020-08-04:
I had to unload the bcm2835_v4l2 bcm2835_isp bcm2835_codec modules before I could use my UVC webcam. It looks like the discovery code doesn't like the Pi camera. And even if you specify a camera by name it still uses the discovery code and looks for the camera by name instead of grabbing it directly.