sarxos/webcam-capture #757: MAC OS - No webcams found

JakobMaaloe, 2020-01-29:
Hi,

I've been using the framework on an old Macbook Pro, but after changing to a new Macbook Pro running Catalina, the framework no longer finds the connected webcams.

There are currently two working webcams (one internal) connected to the Macbook Pro, both have been verified using Google Hangout, etc.

Mac OS: Catalina Version 10.15.2
Computer: i9, 32 GB

Code that fails:
 public BufferedImage takePicture() throws IOException {
        List<Webcam> webcams = Webcam.getWebcams();
        log.info("Found {} webcams", webcams.size());
....

Log:
2020-01-29 09:46:38.152  INFO 5638 --- [nio-8080-exec-7] d.d.a.a.service.CameraService            : Found 0 webcams
2020-01-29 09:46:38.153  WARN 5638 --- [nio-8080-exec-7] com.github.sarxos.webcam.Webcam          : No webcam has been detected!

Any chance something has changed?

Br.
Jakob Maaloe

kkieffer, 2020-02-07:
I think this is a duplicate of #723

sarxos, 2020-03-30:
This should already be fixed by commit https://github.com/sarxos/webcam-capture/commit/ae28f268f081f2a1b13f06578a0f4fddb6ffc568 from @kkieffer.

@JakobMaaloe can you please verify this issue with latest 0.3.13-SNAPSHOT which is available [here](https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture&v=0.3.13-SNAPSHOT)?

JakobMaaloe, 2020-03-31:
Hi,

No luck I'm afraid. Just downloaded the 0.3.13 snapshot, imported it and ran the code again. Initially it gave me a ClassNotFoundException (org.bridj.cpp.CPPObject).

But after including the bridj jar I still get an error (see below). What am I missing here?

java.lang.UnsatisfiedLinkError: 'org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices()'
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices(Native Method) ~[webcam-capture-0.3.13-20200330.202351-7.jar:na]
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.handle(WebcamDefaultDriver.java:93) ~[webcam-capture-0.3.13-20200330.202351-7.jar:na]
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ~[webcam-capture-0.3.13-20200330.202351-7.jar:na]
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) ~[na:na]
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) ~[na:na]
	at java.base/java.lang.Thread.run(Thread.java:834) ~[na:na]

sarxos, 2020-04-01:
@JakobMaaloe Thank you. It seems like we did the test after fix in a wrong way so, and I'm sorry for that, this problem is not yet fixed :(

JakobMaaloe, 2020-04-02:
Ok, no problem. I'll be waiting. :-)

thanosa75, 2020-04-04:
Hello,
just arrived at this point - used 0.3.13-SNAPSHOT; I get this:

```
Caused by: java.lang.UnsatisfiedLinkError: org.openimaj.video.capture.OpenIMAJGrabber.getVideoDevices()Lorg/bridj/Pointer;
	at org.openimaj.video.capture.OpenIMAJGrabber.getVideoDevices(Native Method) ~[core-video-capture-1.3.jar:na]
	at org.openimaj.video.capture.VideoCapture.getVideoDevices(VideoCapture.java:221) ~[core-video-capture-1.3.jar:na]
	at com.github.sarxos.webcam.ds.openimaj.OpenImajDriver$GetDevicesTask.handle(OpenImajDriver.java:58) ~[webcam-capture-driver-openimaj-0.3.13-20200330.202457-6.jar:na]
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ~[webcam-capture-0.3.13-20200330.202351-7.jar:na]
	... 3 common frames omitted
```

and running macOS 10.15.4. No prompt to authorize camera. Any news yet?

kkieffer, 2020-04-04:
@thanosa75 you need an updated lib - see end of thread on #723 still working issue

thanosa75, 2020-04-04:
Thanks for the quick reply @kkieffer !

Got to the bottom of this, going through OpenIMAJ tickets and a few from this repo. Simpler solution that WFM 💯  is:

1. include latest snapshot (although may not make a difference) of the artifacts (as in my comment above)
2. force use of the bridj artifact as such:

a. add repository to pom:
``` 
        <repository>
            <id>bridjhack</id>
            <url>http://maven.ecs.soton.ac.uk/content/repositories/thirdparty/</url>
        </repository>
```

b. add artifacts (no need to exclude transitive one):

```
         <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture-driver-openimaj</artifactId>
            <version>0.3.13-SNAPSHOT</version>
            <exclusions>
                <exclusion>
                    <groupId>org.openimaj</groupId>
                    <artifactId>core-video-capture</artifactId>
                </exclusion>
            </exclusions>
        </dependency>
        <dependency>
            <groupId>com.nativelibs4java</groupId>
            <artifactId>bridj</artifactId>
            <version>0.7-20140918-2</version>
        </dependency>
        <dependency>
            <groupId>org.openimaj</groupId>
            <artifactId>core-video-capture</artifactId>
            <version>1.3.10</version>

            <!-- webcam-capture doesn't need most of the transitive dependencies -->
            <exclusions>
                <exclusion>
                    <artifactId>azure-bing-search-java</artifactId>
                    <groupId>net.billylieurance.azuresearch</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>snuggletex-core</artifactId>
                    <groupId>uk.ac.ed.ph.snuggletex</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>snuggletex-upconversion</artifactId>
                    <groupId>uk.ac.ed.ph.snuggletex</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>snuggletex-jeuclid</artifactId>
                    <groupId>uk.ac.ed.ph.snuggletex</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>flickrapi</artifactId>
                    <groupId>com.aetrion.flickr</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>jal</artifactId>
                    <groupId>vigna.dsi.unimi.it</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>jama</artifactId>
                    <groupId>jama</groupId>
                </exclusion>
                <exclusion>
[... 70 lines cut ...]
                </exclusion>
                <exclusion>
                    <artifactId>httpclient</artifactId>
                    <groupId>org.apache.httpcomponents</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>jmatio</artifactId>
                    <groupId>net.sourceforge.jmatio</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>BezierUtils</artifactId>
                    <groupId>com.caffeineowl.graphics</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>jai-core</artifactId>
                    <groupId>javax.media</groupId>
                </exclusion>
                <exclusion>
                    <artifactId>jai-codec</artifactId>
                    <groupId>com.sun.media</groupId>
                </exclusion>
            </exclusions>
        </dependency>

```

x21L, 2020-07-31:
@thanosa75 Thank you very much for this. I really appreciate it 😄

rubenqba, 2021-05-20:
The fix doesn't work for me. Any advice on what JDK did you use??

javaspeak, 2021-05-25:
@rubenqba  did you get anything working? - I cannot get any driver working with MacOS Catalina.  I have left my comments in

     https://github.com/sarxos/webcam-capture/issues/723

If you do get anything working i will be so happy if you tell me what you did.

Thanks

rubenqba, 2021-05-25:
@javaspeak I got my sample project working using SNAPSHOT version, here a snippet of my dependencies

javaspeak, 2021-05-27:
@rubenqba 

Thanks for your answer.  When I run it I get:

    Thread 41 Crashed:: Dispatch queue: com.apple.root.default-qos
    0   libsystem_kernel.dylib        	0x00007fff6f00bad6 __abort_with_payload + 10
    1   libsystem_kernel.dylib        	0x00007fff6f00d3df abort_with_payload_wrapper_internal + 80
    2   libsystem_kernel.dylib        	0x00007fff6f00d411 abort_with_payload + 9
    3   com.apple.TCC                 	0x00007fff656de59f __CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 163
    4   com.apple.TCC                 	0x00007fff656dc531 __TCCAccessRequest_block_invoke.114 + 500
    5   com.apple.TCC                 	0x00007fff656dca58 __tccd_send_message_block_invoke + 231
    6   libxpc.dylib                  	0x00007fff6f0f01ff _xpc_connection_reply_callout + 36
    7   libxpc.dylib                  	0x00007fff6f0f0187 _xpc_connection_call_reply_async + 69
    8   libdispatch.dylib             	0x00007fff6ee516c2 _dispatch_client_callout3 + 8
    9   libdispatch.dylib             	0x00007fff6ee6815d _dispatch_mach_msg_async_reply_invoke + 369
    10  libdispatch.dylib             	0x00007fff6ee605f9 _dispatch_kevent_worker_thread + 1316
    11  libsystem_pthread.dylib       	0x00007fff6f0aba85 _pthread_wqthread + 362
    12  libsystem_pthread.dylib       	0x00007fff6f0aab77 start_wqthread + 15

Note the line:

     __CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 163

When you run it do you get a pop up asking if you want to allow the webcam ?  I get no such popup.

Also are your running it in an IDE or did you package your app up into a dmg and sign it, etc, to get it working?

I am just running code in an IDE.

rubenqba, 2021-05-27:
I was running inside the IDE (Intellij in my case) without any issue. Also, I don't remember any popup but in my system, the Intellij already has access to the camera. 
You can check if your IDE is allowed to use the camera in **System Preference > Privacy**; select Camera and on the right side check the apps allowed to use it.

javaspeak, 2021-05-27:
====================================================
SUCCESS getting webcam to work with MacOS Catalina - time to party!
====================================================

For anyone else who has problems starting a webcam with MacOS Catalina I just managed :-)

I am so happy :-)

I was trying to run the webcam code from the STS (Eclipse IDe) and it never popped up a security dialog to enable the camera.  Instead the application always crashed out and in Macs Console app I say a crash error like:

     3   com.apple.TCC                 	0x00007fff656de59f __CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 163

The Solution was to start my STS IDE from the terminal as follows:

    /Applications/SpringToolSuite4_3.app/Contents/MacOS/SpringToolSuite4 

I then got a security popup to allow access to the camera and the webcam worked without error.

When I went to the Security and Privacy settings  for the Camera I now see the Terminal application and a ticked checkbox.  prior to the popup there was no Terminal in the list.

I have included a zip file of the code here so you can see the dependencies in the pom.xml and the Java code.

Note I used JRE 15.0.2 that was bundled as a plugin in my STS IDE.

      /Applications/SpringToolSuite4_3.app/Contents/Eclipse/plugins/org.eclipse.justj.openjdk.hotspot.jre.full.macosx.x86_64_15.0.2.v20210201-0955/jre/lib

[webcam_exp.zip](https://github.com/sarxos/webcam-capture/files/6554306/webcam_exp.zip)

kruinkriebel, 2022-03-09:
> SUCCESS getting webcam to work with MacOS Catalina - time to party!
> 
> For anyone else who has problems starting a webcam with MacOS Catalina I just managed :-)
> 
> I am so happy :-)
> 

Thanks @javaspeak!! 
I've  been using this library in my one and only "home project" and it depends heavily on it (because it's an app that creates timelapse video's from periodically taken pictures, intended for capturing a days worth of clouds passing by). 
I had given up on the project 2 years ago, but somehow gave it a spin again today, see if I could fix things. 
Using your repos and dependencies, I've got the project running again! 

I'm on macOS 11.6.4 (Big Sur), still using jdk1.8.0_73.jdk.

eduramiba, 2022-12-09:
Hi, please try this driver, it should work with recent MacOS versions https://github.com/eduramiba/webcam-capture-driver-native

codaman, 2023-08-30:
@javaspeak . i am trying to implement these dependencies to kotlin compose multiplatfrom.

```
maven {
        url = uri("http://oss.sonatype.org/content/repositories/snapshots")
        isAllowInsecureProtocol = true
    }
    maven {
        url = uri("http://maven.openimaj.org")
        isAllowInsecureProtocol = true
    }

    mavenCentral {
        url = uri("https://maven.ecs.soton.ac.uk/content/groups/maven.openimaj.org")
        isAllowInsecureProtocol = true
    }

implementation ("com.github.sarxos:webcam-capture:0.3.13-SNAPSHOT")
implementation("com.github.sarxos:webcam-capture-driver-openimaj:0.3.13-SNAPSHOT") {
                    exclude(group = "org.openimaj", module = "core-video-capture")
                }
 implementation ("com.nativelibs4java:bridj:0.7-20140918-3")
```

But I got:

org.jetbrains.kotlin.scripting.compiler.plugin.ScriptingCompilerConfigurationComponentRegistrar is not compatible with this version of compiler
error: unable to evaluate script, no scripting plugin loaded