sarxos/webcam-capture #760: Raspberry PI, V4l4j. Low fps, cpu load

qdimka, 2020-02-21:
Hi

I started using webcam-capture with v4l4j-driver on raspberry pi 4. I have some troubles with cpu load and low fps.
Cpu load is over 95%. Fps is around 3-5. 

How I can increase fps and decrease cpu load?

I tried guvcview, which uses v4l2, and it gives me 1920*1080 23-25fps, without cpu load. 

Should I use webcamera with h264 codec onboard?

yuripourre, 2020-02-21:
I had some performance trouble with v4l4j once, then I created the ffmpeg driver. It runs using Java Process, but it is fast and compatible with Raspberry Pi.

Do you mind trying it?
https://github.com/sarxos/webcam-capture/tree/a83703041ea505f411972b614386ab1fbf065ecb/webcam-capture-drivers/driver-ffmpeg-cli

qdimka, 2020-02-22:
@yuripourre thanks for the suggestion. I will try it Monday. 

Is that driver available on maven?

yuripourre, 2020-02-22:
Thanks @qdimka, unfortunately the driver is not available in maven yet, I filled an issue #751 but @sarxos didn't answered.

If you really need a maven artifact I can publish a fork to maven central under my domain (harium.com).

qdimka, 2020-02-23:
@yuripourre if it's not difficult for you, it would be great

yuripourre, 2020-03-02:
@qdimka Sorry for the late reply, it should be available on maven central soon:
```
<dependency>
  <groupId>com.harium</groupId>
  <artifactId>webcam-capture-driver-ffmpeg-cli</artifactId>
  <version>0.3.12</version>
</dependency>
```
https://oss.sonatype.org/service/local/repositories/releases/content/com/harium/webcam-capture-driver-ffmpeg-cli/0.3.12/webcam-capture-driver-ffmpeg-cli-0.3.12.jar