sarxos/webcam-capture #761: Inconsistent library versions reminder.

calvinhkf, 2020-03-02:
Hi. I have implemented a tool to detect library version inconsistencies. Your project have 3 inconsistent libraries.

Take org.apache.httpcomponents:httpclient for example, this library is declared as version 4.5.3 in webcam-capture-drivers/driver-ipcam, 4.1.3 in webcam-capture-addons/webcam-capture-addon-spycam and etc... Such version inconsistencies may cause unnecessary maintenance effort in the long run. For example, if two modules become inter-dependent, library version conflict may happen. It has already become a common issue and hinders development progress. Thus a version harmonization is necessary.

Provided we applied a version harmonization, I calculated the cost it may have to harmonize to all upper versions including an up-to-date one. The cost refers to POM config changes and API invocation changes. Take org.apache.httpcomponents:httpclient for example, if we harmonize all the library versions into 4.5.9. The concern is, how much should the project code adapt to the newer library version. We list an effort table to quantify the harmonization cost. 

The effort table is listed below. It shows the overall harmonization effort by modules. The columns represents the number of library APIs and API calls(NA,NAC), deleted APIs and API calls(NDA,NDAC) as well as modified API and API calls(NMA,NMAC). Modified APIs refers to those APIs whose call graph is not the same as previous version. Take the first row for example, if upgrading the library into version 4.5.9. Given that 1 APIs is used in module webcam-capture-drivers/driver-ipcam, 0 of them is deleted in a recommended version(which will throw a NoMethodFoundError unless re-compiling the project),  1 of them is regarded as modified which could break the former API contract.

|Index|Module|NA(NAC)|NDA(NDAC)|NMA(NMAC)|
|-|-|-|-|-|
|1|webcam-capture-drivers/driver-ipcam|1(1)|0(0)|1(1)|
|2|webcam-capture-addons/webcam-capture-addon-spycam|0(0)|0(0)|0(0)|

Also we provided another table to show the potential files that may be affected due to library API change, which could help to spot the concerned API usage and rerun the test cases. The table is listed below. 

|Module|File|Type|API|
|-|-|-|-|
|webcam-capture-drivers/driver-ipcam|webcam-capture-drivers/driver-ipcam/src/main/java/com/github/sarxos/webcam/ds/ipcam/IpCamAuth.java|modify|org.apache.http.auth.UsernamePasswordCredentials.UsernamePasswordCredentials(java.lang.String,java.lang.String)|

If you are interested, you can have a more complete and detailed report in the attached PDF file.
[sarxos webcam-capture.pdf](https://github.com/sarxos/webcam-capture/files/4272992/sarxos.webcam-capture.pdf)

sarxos, 2020-03-30:
Hi @calvinhkf, thank you for the pointer :) I updated source code to address detected issues in 6fee1466017d28526a01c36a3da55baeceb54247, ad4e39cb3f6a04f38c4894c3251026a2639a73b2 and ed8f4341eee67f4014ffd00c5cbcf8347e8727ad. May I ask how this report is generated? Is this an open source project?

calvinhkf, 2020-03-31:
Thanks! The report is generated based on analysis of project dependencies(as maven does but not totally follow the procedures and output of maven) and extraction of project API usages on dependencies.  The projects is not open-sourced yet. I want to make sure it is really helpful to developers.

sarxos, 2020-04-01:
Thank you @calvinhkf :) In return let me provide some feedback.

In the _Harmonization Effort_ tables it would be good to have only these modules which has NA/NDA/NMA greater than 0. With a greater number of modules the rows where these stats are 0 (zeros) obscure the table and make it harder to read (too long). It would be nicer to have them filtered out so only these which are greater than 0 are kept, IMO.

In the Files Related, the API column I see something suspicious. It states (or maybe I understood it incorrectly) that the following APIs are modified between 0.3.11 and 0.3.12, and thus harmonization effort is required:

* `com.github.sarxos.webcam.Webcam.getDefault()`
* `com.github.sarxos.webcam.Webcam.open()`
* `com.github.sarxos.webcam.Webcam.setViewSize(java.awt.Dimension)`
* `com.github.sarxos.webcam.Webcam.getImage()`
* `com.github.sarxos.webcam.Webcam.close()`

But when I look for example at `Webcam.setViewSize(java.awt.Dimension)` I see no difference in the method signature or the method code between version 0.3.11, 0.3.12 and master branch. So I do not fully understand what the _Type_ column with value _Modified_ really means in this context.

calvinhkf, 2020-04-02:
_Modified_ means the API signature remains the same, but some of its calling methods that are nodes of the API call graph changes. Those _modified API signatures_ are considered as method _addition_ and _deletion_ in our context. The report didn't include a call chain that relates an API entrance to the modified sub methods. And your confusion is due to my briefing report that didn't include the info as mentioned above(If shown on a PDF report, the tables and lists would be too hard to read). 

But I think both of the report's redundancy and deficiency will be further overcome if the result is shown on an interactive GUI such as web(which is currently on a roadmap).

Thanks for your feedback!   Appreciate it very much!

sarxos, 2020-04-02:
Ok, I get it now. I'm closing this ticket as resolved. Thank you!

closed 2020-04-02 by sarxos