sarxos/webcam-capture #764: How to perform QR code analysis while displaying camera pictures

geekeritcom, 2020-03-23:
How to perform QR code analysis while displaying camera pictures，Now I can only display pictures or QR code, but they cannot work at the same time

sarxos, 2020-03-23:
Hi @geekeritcom,

> Now I can only display pictures or QR code, but they cannot work at the same time

This is not true. You can display both image from webcam, and QR code. The problem is how you implement this :)

You can for example open camera in async mode so it does not block when you call `getImage()`, pass it to `WebcamPanel`, and use the very same `Webcam` instance in a separate `Thread` to read QR code from image. Both image display and analysis will be done here in parallel.

Another example would be have a custom `WebcamImageTransformer` which reads the code from image before it's returned from `getImage()`, but this solution will most likely drop effective FPS since QR code analysis time will be added to the `getImage()` execution time. QR code analysis and image display will be done here one after the other (and for every image)

Please take a look at this basic example. Here image from webcam id displayed in parallel to the QR code analysis:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-examples/webcam-capture-qrcode/src/main/java/com/github/sarxos/example1/WebcamQRCodeExample.java

This example is the embodiment of the the first idea I described above.

geekeritcom, 2020-03-24:
Thank you very much for responding so fast. Yesterday I also found a solution by referring to the image of the network camera. I saw your reply code today. Thank you very much. In any case, thank you for your reply and such a great tool.

```
private static void runTask(Webcam webcam) {
        ExecutorService executorService = Executors.newSingleThreadExecutor();
        executorService.submit(() -> {
            webcam.setViewSize(WebcamResolution.VGA.getSize());

            WebcamPanel panel = new WebcamPanel(webcam);

            panel.setFPSDisplayed(true);
            panel.setDisplayDebugInfo(true);
            panel.setImageSizeDisplayed(true);
            panel.setMirrored(true);

            Thread thread = new Thread() {
                @Override
                public void run() {
                    while (true) {
                        if (!webcam.isOpen()) {
                            webcam.open(true);
                        }
                        BufferedImage image = webcam.getImage();
                        try {
                            ZxingUtil.decodeImg(image);
                        } catch (Exception e) {
                            e.printStackTrace();
                        }
                        try {
                            TimeUnit.SECONDS.sleep(1);
                        } catch (InterruptedException e) {
                            e.printStackTrace();
                        }
                    }
                }
            };
            thread.start();

            JFrame window = new JFrame("Test webcam panel");
            window.add(panel);
            window.setResizable(true);
            window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
            window.pack();
            window.setVisible(true);
        });
    }
```

closed 2020-03-26 by geekeritcom