sarxos/webcam-capture #766: UnsatisfiedLinkError against OpenIMAJGrabber.startSession(IIILorg/bridj/Pointer;)

sarxos, 2020-03-30:
Opening `Webcam` cause this error on Linux machine:

```plain
21:02:21.519 [SwingWorker-pool-2-thread-1] DEBUG com.github.sarxos.webcam.Webcam - Webcam exception when opening
com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72) ~[classes/:na]
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140) ~[classes/:na]
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46) ~[classes/:na]
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.open(WebcamOpenTask.java:20) ~[classes/:na]
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:295) [classes/:na]
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:250) [classes/:na]
	at com.github.sarxos.webcam.Webcam.open(Webcam.java:236) [classes/:na]
	at com.github.sarxos.webcam.WebcamPanel$1.doInBackground(WebcamPanel.java:840) [classes/:na]
	at com.github.sarxos.webcam.WebcamPanel$1.doInBackground(WebcamPanel.java:1) [classes/:na]
	at javax.swing.SwingWorker$1.call(SwingWorker.java:295) [na:1.8.0_201]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) [na:1.8.0_201]
	at javax.swing.SwingWorker.run(SwingWorker.java:334) [na:1.8.0_201]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149) [na:1.8.0_201]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624) [na:1.8.0_201]
	at java.lang.Thread.run(Thread.java:748) [na:1.8.0_201]
Caused by: java.lang.UnsatisfiedLinkError: com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.startSession(IIILorg/bridj/Pointer;)Z
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.startSession(Native Method) ~[classes/:na]
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice.open(WebcamDefaultDevice.java:332) ~[classes/:na]
	at com.github.sarxos.webcam.ds.cgt.WebcamOpenTask.handle(WebcamOpenTask.java:38) ~[classes/:na]
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ~[classes/:na]
	... 3 common frames omitted
```

sarxos, 2020-03-30:
Hi @kkieffer

This issue is caused by a method parameter signature being changed from `double` to `int` in your commit here:

https://github.com/sarxos/webcam-capture/pull/765/files#diff-492ecf2a3cddaf20cf2f0e1a230fa31d.

I initially thought it's a valid change since it's only normal to assume that FPS is `int`, but it looks like BridJ needs it to be double to link correct function exported in native library, thus I'm reverting it back to how it was before.

Can you please test newest `0.3.13-SNAPSHOT` on your MacOS machine to verify if this change works well? I have no access to Apple boxes and can't verify it.

kkieffer, 2020-03-30:
It won't compile correctly if it's a double.  The native code wants an int.  See the native source here, where reqMillisPerFrame is an int:

[image]

sarxos, 2020-03-30:
@kkieffer 

Ok, I understand now why you had to change it. Let me check why it's not linking on my Linux box. I suppose there is a bug somewhere else in the code or pom since OpenIMAJ also have `int` in `OpenIMAJGrabber` class.

sarxos, 2020-03-30:
@kkieffer 

Nailed it. The `so` files for x64 and x86 Linux are outdated (others probably as well). I will update them.

sarxos, 2020-03-30:
@kkieffer Thank you for your help. Everything should be working now. Can I ask you one more time to verify if the 0.3.13-SNAPSHOT is working well on Mac? I suppose it should be working but better be safe than sorry :) The latest snapshot is already deployed in Sonatype and available [here](https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture&v=0.3.13-SNAPSHOT).

kkieffer, 2020-03-30:
Working!

sarxos, 2020-04-01:
Great. Thank you!