sarxos/webcam-capture #777: Inclusion into IntelIJ

Abdus-Samee, 2020-06-24:
I am having the following errors. Is there any linking of libraries I haven't done?

Error:(44, 47) java: package ch.qos.logback.classic does not exist
Error:(44, 95) java: package ch.qos.logback.classic does not exist
Error:(45, 53) java: package ch.qos.logback.classic.joran does not exist
Error:(45, 119) java: package ch.qos.logback.classic.joran does not exist
Error:(293, 66) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver
Error:(328, 75) java: incompatible types: com.github.sarxos.webcam.WebcamUpdater.DelayCalculator cannot be converted to webcamsrc.com.github.sarxos.webcam.WebcamUpdater.DelayCalculator
Error:(370, 68) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver
Error:(458, 64) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver
Error:(661, 80) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver
Error:(730, 64) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver
Error:(774, 58) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver
Error:(1045, 34) java: incompatible types: com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver cannot be converted to webcamsrc.com.github.sarxos.webcam.WebcamDriver
Error:(161, 74) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver
Error:(224, 75) java: incompatible types: webcamsrc.com.github.sarxos.webcam.WebcamDriver cannot be converted to com.github.sarxos.webcam.WebcamDriver

shinobisoft, 2020-09-22:
Why aren't you using a prebuilt webcam-capture jar? That's a major part of the log you shared...

All but the first 4 errors are from your code.   "webcamsrc.com.github.sarxos.webcam.xxx" should be "com.github.sarxos.webcam.xxx". This tells me you're attempting to build webcam-capture locally but you've renamed the entire package by adding an additional package named "webcamsrc" to your project structure. Removing this first "package" should fix most, if not all, of your build errors.

sarxos, 2021-02-02:
@Abdus-Samee Your packages are obviously messed up, as @shinobisoft mentioned above. I'm closing this issue.

closed 2021-02-02 by sarxos