sarxos/webcam-capture #782: Can't execute on armfh 64 bits

megahertzon, 2020-08-31:
Hello I want to execute this example:
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamPanel;
import com.github.sarxos.webcam.WebcamResolution;

public class WebcamPanelExample {

	public static void main(String[] args) throws InterruptedException {

		Webcam webcam = Webcam.getDefault();
		webcam.setViewSize(WebcamResolution.VGA.getSize());

		WebcamPanel panel = new WebcamPanel(webcam);
		panel.setFPSDisplayed(true);
		panel.setDisplayDebugInfo(true);
		panel.setImageSizeDisplayed(true);
		panel.setMirrored(true);

		JFrame window = new JFrame("Test webcam panel");
		window.add(panel);
		window.setResizable(true);
		window.setDefaultCloseOperation(JFrame.EXIT_ON_CLOSE);
		window.pack();
		window.setVisible(true);
	}
}

it works fine on mac 64 bit operating system.

I can´t execute on armfh SBC 64 bit, the problem:

java.lang.RuntimeException: Failed to initialize BridJ (java.lang.UnsatisfiedLinkError: /tmp/BridJExtractedLibraries9042634838982220816/libbridj.so: /tmp/BridJExtractedLibraries9042634838982220816/libbridj.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform))
	at org.bridj.Platform.initLibrary(Platform.java:444)
	at org.bridj.Platform.<clinit>(Platform.java:228)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:57)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.UnsatisfiedLinkError: /tmp/BridJExtractedLibraries9042634838982220816/libbridj.so: /tmp/BridJExtractedLibraries9042634838982220816/libbridj.so: cannot open shared object file: No such file or directory (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform)
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1934)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1817)
	at java.lang.Runtime.load0(Runtime.java:810)
	at java.lang.System.load(System.java:1088)
	at org.bridj.Platform.initLibrary(Platform.java:420)
	... 7 more

is there available 64 bit armfh libraries??

Best regards, Nelson

gabrielandrade2, 2020-10-26:
I have a similar issue when running on a aarch64 platform (Snapdragon 410c).

```
java -jar csitest.jar                                                            
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
java.lang.RuntimeException: Failed to initialize BridJ (java.lang.UnsatisfiedLinkError: /tmp/BridJExtractedLibraries2)
        at org.bridj.Platform.initLibrary(Platform.java:447)
        at org.bridj.Platform.<clinit>(Platform.java:231)
        at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.ja)
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.UnsatisfiedLinkError: /tmp/BridJExtractedLibraries2571491173185175723/libbridj.so: /tmp/BridJExt)
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
        at java.lang.Runtime.load0(Runtime.java:809)
        at java.lang.System.load(System.java:1086)
        at org.bridj.Platform.initLibrary(Platform.java:423)
        ... 7 more
Exception in thread "main" java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.lang.reflect.Method.invoke(Method.java:498)
        at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcak
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:866)
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:845)
        at Main.main(Main.java:36)
        ... 5 more
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at java.util.concurrent.FutureTask.report(FutureTask.java:122)
        at java.util.concurrent.FutureTask.get(FutureTask.java:192)
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
        ... 9 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDrive)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:117)
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.UnsatisfiedLinkError: org.bridj.Platform.sizeOf_ptrdiff_t()I
        at org.bridj.Platform.sizeOf_ptrdiff_t(Native Method)
        at org.bridj.Platform.<clinit>(Platform.java:235)
        at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.ja)
```

theHilikus, 2020-12-09:
Did you guys find a solution? i'm having the same issue on an aarch64 board running linux