sarxos/webcam-capture #792: atomic-processor-1 allocated bytes grows until system is unstable

derpnooner, 2020-11-17:
I'm trying to create a very simple app that will either display an error message or display an image from the USB camera.  The application works fine, but it needs to be able to run indefinitely.  The problem is the system becomes completely unstable after about an hour and looking at the process in TOP, I see the MEM% go upwards to 90% before it crashes.

I'm experiencing a memory leak calling the Webcam.open / webcam.getImage() / webcam.close -- looking at VisualVM, the thread for atomic-processor-1 allocated bytes just continually grows until the entire machine crashes.

No matter what I attempt, I cannot get it to release any memory and I'm not finding any issues similar.  Please help!

Main Class:
`package webcamCapture;

import java.awt.EventQueue;
import javax.swing.JFrame;
import javax.swing.ImageIcon;
import javax.swing.JButton;
import java.awt.event.ActionListener;
import java.io.IOException;
import java.io.OutputStream;
import java.io.PrintStream;
import java.awt.event.ActionEvent;
import java.awt.Canvas;
import java.awt.Font;
import java.awt.Graphics;

import javax.swing.JPanel;
import javax.swing.border.EtchedBorder;
import javax.swing.text.DefaultCaret;

import javax.swing.JTextArea;
import java.awt.BorderLayout;
import javax.swing.JScrollPane;
import javax.swing.JLabel;
import javax.swing.SwingConstants;

public class WebcamCapture_gui {
	
	static JFrame frmWebcamCapture_gui;
	static Canvas canvas = new Canvas();
	static Graphics g;
	private static JTextArea textArea;
	private static Thread t1;
	private static boolean running = true;
	
	
	
	
	public static void main(String[] args) {
		
		t1 = new Thread(new thread1());
		
		EventQueue.invokeLater(new Runnable() {
			public void run() {
				try {
					@SuppressWarnings("unused")
					WebcamCapture_gui window = new WebcamCapture_gui();
					//window.frmWebcamCapture_gui.setVisible(true);
				} catch (Exception e) {
					e.printStackTrace();
				}
			}
			
			
		});
		
	}
	
	public WebcamCapture_gui() throws IOException {
		
		initialize();
		new Thread(t1).start();
	}

[... 287 lines cut ...]
				webcam.close();
				
				webcam = null;
				
				WebcamCapture_gui.g.drawImage(camImage, 0, 0, 1048, 543, null);
				
				camImage.flush();
				camImage = null;
				
				WebcamCapture_gui.g.dispose();
				
			} else {
				
				loadNotFound();
				
				WebcamCapture_gui.g.drawImage(notFound, 0, 0, 1048, 543, null);
				
				notFound.flush();
				notFound = null;
				
				WebcamCapture_gui.g.dispose();
			}
	    });
	}
}`

alexmao86, 2021-02-03:
working on...

jeremylcarter, 2026-01-13:
Any update on this?