sarxos/webcam-capture #794: Library 'OpenIMAJGrabber' was not found in path

hatemibr, 2020-11-22:
Hello there,

I have been trying to find solution in all over the internet. I deeply investigated every ticket about this error but i have lost my day doing that. I am lost I NEED your help:

I ' am using the last stable version of webcam (webcam-capture-0.3.12) - and I have successfully added the jar files to my program and i just tried to add this code 

```
Webcam webcam = Webcam.getDefault();
		webcam.open();
```

but I am getting a very strange Exception :

```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
java.io.FileNotFoundException: Library 'OpenIMAJGrabber' was not found in path '[., C:\Program Files\Common Files\Oracle\Java\javapath, C:\Windows\system32, C:\Windows, C:\Windows\System32\Wbem, C:\Windows\System32\WindowsPowerShell\v1.0\, C:\Windows\System32\OpenSSH\, C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common, C:\Users\Hatem's Pc\AppData\Local\Microsoft\WindowsApps, C:\Program Files\Java\jdk-15.0.1\bin, C:\Windows\system32, C:\Windows, C:\Program Files\Common Files\Oracle\Java\javapath, C:\Windows\system32, C:\Windows, C:\Windows\System32\Wbem, C:\Windows\System32\WindowsPowerShell\v1.0\, C:\Windows\System32\OpenSSH\, C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common, C:\Users\Hatem's Pc\AppData\Local\Microsoft\WindowsApps, ., C:\Program Files\Java\jdk-15.0.1\bin, C:\Program Files\Java\jdk-15.0.1\bin]'
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:1074)
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:1049)
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:602)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:771)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:1022)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:904)
	at bridj@0.7.0/org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:271)
	at bridj@0.7.0/org.bridj.BridJ.initialize(BridJ.java:1128)
	at bridj@0.7.0/org.bridj.NativeObject.<init>(NativeObject.java:50)
	at bridj@0.7.0/org.bridj.StructObject.<init>(StructObject.java:46)
	at bridj@0.7.0/org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:64)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
	at java.base/java.lang.Thread.run(Thread.java:832)
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:956)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:933)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:911)
	at client.PictureTaker.getPicture(PictureTaker.java:19)
	at Engine.Engine.main(Engine.java:29)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.base/java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.base/java.util.concurrent.FutureTask.get(FutureTask.java:191)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 6 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:45)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:117)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1130)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:630)
	at java.base/java.lang.Thread.run(Thread.java:832)
Caused by: java.lang.RuntimeException: Failed to allocate new instance of type class com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:812)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:1022)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:904)
	at bridj@0.7.0/org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:271)
	at bridj@0.7.0/org.bridj.BridJ.initialize(BridJ.java:1128)
	at bridj@0.7.0/org.bridj.NativeObject.<init>(NativeObject.java:50)
	at bridj@0.7.0/org.bridj.StructObject.<init>(StructObject.java:46)
	at bridj@0.7.0/org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:64)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at webcam.capture@0.3.12/com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more
Caused by: java.io.FileNotFoundException: Library 'OpenIMAJGrabber' was not found in path '[., C:\Program Files\Common Files\Oracle\Java\javapath, C:\Windows\system32, C:\Windows, C:\Windows\System32\Wbem, C:\Windows\System32\WindowsPowerShell\v1.0\, C:\Windows\System32\OpenSSH\, C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common, C:\Users\Hatem's Pc\AppData\Local\Microsoft\WindowsApps, C:\Program Files\Java\jdk-15.0.1\bin, C:\Windows\system32, C:\Windows, C:\Program Files\Common Files\Oracle\Java\javapath, C:\Windows\system32, C:\Windows, C:\Windows\System32\Wbem, C:\Windows\System32\WindowsPowerShell\v1.0\, C:\Windows\System32\OpenSSH\, C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common, C:\Users\Hatem's Pc\AppData\Local\Microsoft\WindowsApps, ., C:\Program Files\Java\jdk-15.0.1\bin, C:\Program Files\Java\jdk-15.0.1\bin]'
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:1074)
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:1049)
	at bridj@0.7.0/org.bridj.BridJ.getNativeLibrary(BridJ.java:602)
	at bridj@0.7.0/org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:771)
	... 13 more
```

any ideas what's going on here ?
I am using Windows 10 64bit

hatemibr, 2020-11-22:
@sarxos can I get your attention on this bug please ?

Moimus, 2020-11-23:
+1 I have nearly the same issue.

hatemibr, 2020-11-23:
You said it works fine on win10 64bit? 

can you please provide the steps you followed in order to let it works on win10 ? 
Step by step

DanielMartensson, 2020-12-25:
I have the same issue with Raspberry Pi 4B. I can't use the camera. I have open an issue.

hatemibr, 2021-03-03:
This issue has been fixed by moving **"OpenIMAJGrabber.dll"** library to C:\Windows\System32.

This library can be found into the documents which the author developed. (src\com\github\sarxos\webcam\ds\buildin\lib\win64)

closed 2021-03-03 by hatemibr