sarxos/webcam-capture #796: Catch the SampleCB error

Kybox, 2020-12-14:
Hi !
Could someone tell me how to catch the error from the log :
`ERROR: SampleCB() - buffer sizes do not match`

It's just a System.out.println log and I have no class targeted to understand the error.
It appends when I select the OBS device not launched.

Thanks.

alexmao86, 2021-01-30:
looks like this is not from java side

sarxos, 2021-02-02:
Hello there. Yes, it's not from Java. It comes from the underlying native library which uses `videoInput`. The `videoInput` seems to print this error in here:

https://github.com/ofTheo/videoInput/blob/7532fd6ce204ca5e7e7b88c994d453c8831515b9/videoInputSrcAndDemos/libs/videoInput/videoInput.cpp#L189

But it was not updated for ages and I'm not sure how to fix it.

closed 2021-02-02 by sarxos

Toshimichi0915, 2022-04-21:
I'm facing this issue. Has anyone solve this issue?