sarxos/webcam-capture #799: Could Not Initialize OpenIMAJGrabber

jhess32259, 2020-12-29:
Greetings, @sarxos.

I'm receiving an exception when instantiating a JPanel form. Environment is NetBeans 12.1 with OpenJDK15. Below is the code to the JPanel, the exception text, and the POM. Any help would be greatly appreciated.

Regards,
Joe Hess

***JPanel Code***

package sst01;

import com.github.sarxos.webcam.Webcam;

public class WebCamera extends javax.swing.JPanel {

    private Webcam webcam;
    
    public WebCamera() {
        initComponents();
        
        webcam = Webcam.getDefault();        
    }

    /**
     * This method is called from within the constructor to initialize the form.
     * WARNING: Do NOT modify this code. The content of this method is always
     * regenerated by the Form Editor.
     */
    @SuppressWarnings("unchecked")
    // <editor-fold defaultstate="collapsed" desc="Generated Code">                          
    private void initComponents() {

        setMaximumSize(new java.awt.Dimension(640, 480));
        setMinimumSize(new java.awt.Dimension(640, 480));
        setPreferredSize(new java.awt.Dimension(640, 480));

        javax.swing.GroupLayout layout = new javax.swing.GroupLayout(this);
        this.setLayout(layout);
        layout.setHorizontalGroup(
            layout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
            .addGap(0, 640, Short.MAX_VALUE)
        );
        layout.setVerticalGroup(
            layout.createParallelGroup(javax.swing.GroupLayout.Alignment.LEADING)
            .addGap(0, 480, Short.MAX_VALUE)
        );
    }// </editor-fold>                        

    // Variables declaration - do not modify                     
    // End of variables declaration                   
}

***Exception***
com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:956)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:933)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:911)
	at sst01.WebCamera.<init>(WebCamera.java:25)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:64)
	at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.base/java.lang.reflect.Constructor.newInstanceWithCaller(Constructor.java:500)
	at java.base/java.lang.reflect.ReflectAccess.newInstance(ReflectAccess.java:128)
	at java.base/jdk.internal.reflect.ReflectionFactory.newInstance(ReflectionFactory.java:350)
	at java.base/java.lang.Class.newInstance(Class.java:645)
	at org.netbeans.modules.form.CreationFactory.createDefaultInstance(CreationFactory.java:155)
	at org.netbeans.modules.form.RADComponent.createBeanInstance(RADComponent.java:227)
	at org.netbeans.modules.form.RADComponent.initInstance(RADComponent.java:166)
[... 170 lines cut ...]
        
        <dependency>
            <groupId>org.swinglabs</groupId>
            <artifactId>swing-layout</artifactId>
            <version>1.0.3</version>
        </dependency>
        
        <dependency>
            <groupId>org.jdesktop</groupId>
            <artifactId>appframework</artifactId>
            <version>1.0.3</version>
        </dependency>
        
        <dependency>
            <groupId>org.eclipse.jetty</groupId>
            <artifactId>jetty-webapp</artifactId>
            <version>9.4.31.v20200723</version>
        </dependency>
        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture</artifactId>
            <version>0.3.12</version>
        </dependency>
    </dependencies>
</project>

farshidmh, 2020-12-30:
I'm getting the exact ( almost ) same error on my logger server

I can't replicate this on my machine  but some of my clients are getting this
POM: webcam-capture 0.3.12

>   [ERROR] dcontroller.IntroController:56 - Launch Webcam Error:#015
      at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
      at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
      at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:866)
      at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:845)
      at dcontroller.IntroController.nextBtnClick(IntroController.java:28)
      at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
      at sun.reflect.NativeMethodAccessorImpl.invoke(Unknown Source)
      at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
      at sun.reflect.misc.Trampoline.invoke(Unknown Source)
      at java.lang.reflect.Method.invoke(Unknown Source)
      at sun.reflect.GeneratedMethodAccessor1.invoke(Unknown Source)
      at sun.reflect.DelegatingMethodAccessorImpl.invoke(Unknown Source)
      at java.lang.reflect.Method.invoke(Unknown Source)
      at sun.reflect.misc.MethodUtil.invoke(Unknown Source)
      at javafx.fxml.FXMLLoader$MethodHandler.invoke(FXMLLoader.java:1769)
      at javafx.fxml.FXMLLoader$ControllerMethodEventHandler.handle(FXMLLoader.java:1657)
      at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:86)
      at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
      at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
      at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
      at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
      at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
      at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
      at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
      at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
      at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
      at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
      at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:89)
      at com.sun.javafx.event.CompositeEventHandler$NormalEventHandlerRecord.handleBubblingEvent(CompositeEventHandler.java:218)
      at com.sun.javafx.event.CompositeEventHandler.dispatchBubblingEvent(CompositeEventHandler.java:80)
      at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:238)
      at com.sun.javafx.event.EventHandlerManager.dispatchBubblingEvent(EventHandlerManager.java:191)
      at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:58)
      at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:49)
      at javafx.event.Event.fireEvent(Event.java:198)
      at javafx.scene.Node.fireEvent(Node.java:8411)
      at javafx.scene.control.Button.fire(Button.java:185)
      at com.sun.javafx.scene.control.skin.BehaviorSkinBase$1.handle(BehaviorSkinBase.java:96)
      at com.sun.javafx.scene.control.behavior.ButtonBehavior.mouseReleased(ButtonBehavior.java:182)
      at com.sun.javafx.event.CompositeEventDispatcher.dispatchBubblingEvent(CompositeEventDispatcher.java:59)
      at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
      at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
      at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
      at com.sun.javafx.event.BasicEventDispatcher.dispatchEvent(BasicEventDispatcher.java:56)
      at com.sun.javafx.event.EventDispatchChainImpl.dispatchEvent(EventDispatchChainImpl.java:114)
      at java.security.AccessController.doPrivileged(Native Method)
      at com.sun.javafx.tk.quantum.GlassViewEventHandler.lambda$handleMouseEvent$2(GlassViewEventHandler.java:432)
      at com.sun.javafx.event.EventUtil.fireEventImpl(EventUtil.java:74)
      at com.sun.javafx.event.EventUtil.fireEvent(EventUtil.java:54)
      at javafx.event.Event.fireEvent(Event.java:198)
      at javafx.scene.Scene$MouseHandler.process(Scene.java:3757)
      at javafx.scene.Scene$MouseHandler.access$1500(Scene.java:3485)
      at javafx.scene.Scene.impl_processMouseEvent(Scene.java:1762)
      at javafx.scene.Scene$ScenePeerListener.mouseEvent(Scene.java:2494)
      at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:394)
      at com.sun.javafx.tk.quantum.GlassViewEventHandler$MouseEventNotification.run(GlassViewEventHandler.java:295)
      at com.sun.javafx.tk.quantum.QuantumToolkit.runWithoutRenderLock(QuantumToolkit.java:410)
      at com.sun.javafx.tk.quantum.GlassViewEventHandler.handleMouseEvent(GlassViewEventHandler.java:431)
      at com.sun.glass.ui.View.handleMouseEvent(View.java:555)
      at com.sun.glass.ui.View.notifyMouse(View.java:937)
      at com.sun.glass.ui.win.WinApplication._runLoop(Native Method)
      at java.lang.Thread.run(Unknown Source)
[... cut ...]
      at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
      at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
      ... 3 more
      at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:45)
      at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:117)
 Caused by: java.lang.NoClassDefFoundError: Could not initialize class com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber

alexmao86, 2021-01-30:
Hi @farshidmh,
I am not sure what is your OS, linux x86 arm? 
it looks OpenIMAJ precompiled native. i.e. JNI is not supporred by your OS. natives are not resolved.

com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber

farshidmh, 2021-01-30:
@alexmao86 

users OS is mostly Windows 7,8 or 10 and the application is delivered with bundled JRE (1.8)

alexmao86, 2021-01-30:
Hi @farshidmh,
The biggest challenge is reproduce the issue. For windows, openIMAG definitely supported, besides windows, linux, arm, armhf...are supported.

"Caused by: java.lang.NoClassDefFoundError: Could not initialize class" no means class not in classpath, really reason is the static block of lading JNI failed.

so questions are:

-  which bridj version and openimaj you are using?
- Are you using maven or gradle for your javafx or download jars by yourself?

regarding stops for long time running, a bells ring in head is that we will review if any thread safe issue that open/close session not in pair causing potential leak issue. we will check.

farshidmh, 2021-01-30:
Hi @alexmao86 

I'm using maven 0.3.12, it will install bridj 0.7.0

`
<dependency>
<groupId>com.github.sarxos</groupId>
<artifactId>webcam-capture</artifactId>
<version>0.3.12</version>
        </dependency>
`

As I mentioned we are using 1.8 thus javafx is builtin