sarxos/webcam-capture #801: how to select video input channel when used in video card with more than one input .

schdai, 2021-01-05:
Hi, I'm using video capture card to get image from some devices. There are more than one input, including  S-Video, Composite and etc.  Can webCam select input channel？ How？

TKS。

alexmao86, 2021-02-02:
use WebcamCompositeDriver

sarxos, 2021-02-02:
@schdai 

The drivers from WebcamCapture project supports only UVC devices and so will not be able to detect cameras connected via S-Video or Composite Cinch, but in general, if you implement (create) custom driver for every input you desire, then you can bundle these drivers together using the `WebcamCompositeDriver` as @alexmao86 suggested.

Just for your information. I never found a Java project to stream video from S-Video or Composite Cinch. I guess you will need some kind of TV tuner for this. Not sure, I never worked with analog video data.

closed 2021-02-02 by sarxos