sarxos/webcam-capture #802: Webcam resolutions missing

Maxforce12, 2021-01-11:
I currently am using [this webcam](https://www.amazon.com/Microphone-Unzano-Computer-Streaming-Conference/dp/B08GPZK2ZQ) to capture and display an image on a JavaFX application.  Everything works fine until I try to change the resolution to 1920x1080, in which case no error is thrown but the code stops running.  I figured I should check and make sure the FHD viewsize is within the webcam.getViewSizes() list, but it appears that the view sizes only go up to 640x480.  Here is my code to display the viewsizes 
`for (Webcam webcam : Webcam.getWebcams()) {
					if (webcam.getName().equalsIgnoreCase(webcamName)) {
						webCam = webcam;
					}
				}` 
and here is my code when I attempt to change the viewsize to 1080p
 `webCam.setViewSize(new Dimension(1920,1080));`
and finally here are the viewsizes that the program says my webcam accepts
 `java.awt.Dimension[width=176,height=144]
java.awt.Dimension[width=320,height=240]
java.awt.Dimension[width=640,height=480]
java.awt.Dimension[width=176,height=144]
java.awt.Dimension[width=320,height=240]
java.awt.Dimension[width=640,height=480]`  

Surely other resolutions should be supported by my webcam.  I even tested it on other applications and it seemed to run fine at 1920x1080
Would anyone know how I might be able to fix this issue?

alexmao86, 2021-02-01:
Hi Maxforce,
(1920,1080) is not listed in default built-in driver , please check enum WebcamResolution java class.

By default failOnSizeMismatch is false, so you set to (1920, 1080) forcely will not throw exception.

If you are sure 1920x1080 is supported, the conclusion is that bottom layer openIMAG native c++ does not detected 1920x1080 on your device. This is  kindly out of webcam scope.

Please consider other drivers. such as OpenCV. if you are on windows, there is one high performance driver https://github.com/alexmao86javafork/webcam-capture-driver-native, you can try it.

thanks,

sarxos, 2021-02-02:
@Maxforce12

Yes, as @alexmao86 suggested you can try switching to the mentioned driver on Windows so all the supported resolutions can be detected automatically.

If not, you may also try doing this in the following way (handwired, no guarantee to work, though):

```java
webcam.setCustomViewSizes(new Dimension(1920,1080)); // register custom resolutions
webcam.setViewSize(new Dimension(1920,1080));
```

For more details please see the code here:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/main/java/com/github/sarxos/webcam/Webcam.java#L549

And the example here:

https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/example/java/CustomResolutionExample.java

closed 2021-02-02 by sarxos