sarxos/webcam-capture #804: Dependencies cannot be resolved

timothyPdevries, 2021-01-22:
There should be no dependencies. Logging should not cause the failure of a project. Sort it out. System.out.println. System.err.println. How hard is that?

Instead I'm messing around with dependencies that don't work, when I should be able to just include the jar. Period. Full stop.

Include the jar and it works. No? Then it's not worth my time. I'll decompile and remove dependencies. Maybe republish as my own work sans dependencies.

eduramiba, 2021-01-22:
Everybody uses slf4j, it's normal. I advice you try using maven even though it may seem unnecesary now.

closed 2021-01-23 by timothyPdevries

timothyPdevries, 2021-01-23:
Thanks, I was able to get it working. I guess I must have been downloading the wrong jar. I read part of the documentation and downloaded a suitable jar. It might be worthwhile having the jars and links to them posted to save the user the confusion and to have the most applicable jars installed.