sarxos/webcam-capture #805: Is this project abondoned?

farshidmh, 2021-01-28:
Hello,

Is this project abandoned?
if so, does anyone know a maintained one?

sarxos, 2021-01-28:
Hi @farshidmh 

This project is not abandoned but it's hard for maintainer to maintain it due to his other activities. The code is open source and not really complicated so I encourage everyone to take a look and fix the bugs, if there are any. Pull requests are always welcome!

farshidmh, 2021-01-29:
Hi @sarxos 

I'm only asking because there are many open issues here with no answer.

The core of my project is based  on this library and I'm already facing some problems that are mentioned here without any answers

sarxos, 2021-01-29:
@farshidmh I suppose you mean the one where you are involved, that is the #799? This one is unfortunate because it requires deeper debugging on the machine where the problem appears. Many of these issues is left open because I'm unable to reproduce them.

farshidmh, 2021-01-29:
@sarxos  exactly,
I asked my client to bring his machine to me!! (the only solution that comes to my mind)
I'll try to come up with a workaround for this and send a pull request.

sarxos, 2021-02-02:
@farshidmh

Did you try the SNAPSHOT version? It's available from the Sonatype repository ([link](https://oss.sonatype.org/service/local/artifact/maven/redirect?r=snapshots&g=com.github.sarxos&a=webcam-capture&v=0.3.13-SNAPSHOT)).

If it's loading on one machine but not on the other it may be an issue in BridJ. It's unfortunate that BridJ is no longer maintained or the maintainer has no time to work on it any more. You can also try adding `http://maven.openimaj.org` as the Maven repository in your POM and force using newer BridJ version, namely `0.7-20140918-3`. This version is crafted by the community, not the maintainer, so it's not available in Maven Central. This is why repository need to be added in POM.

farshidmh, 2021-02-06:
@sarxos 

I'll try this, thank you.

I came up with a workaround, for now, I added a Thread.Sleep to the grabber while 
It'll lower the fps but for some reason, it is working right now

closed 2021-05-31 by alexmao86