sarxos/webcam-capture #808: No webcam detected

djovla, 2021-02-24:
i used the code below and when executed it does not detect any webcam i am using Mac OS and java version 15

Webcam  webcam -= Webcam.getDefault();
if (webcam != null){
System.out.println("Webcam :"+webcam.getName());
}
else
{
System.out.println("No webcam detected");
}

shinobisoft, 2021-03-02:
If this is indeed the code you're using, there is an error with your variable assignment. 
 
[code]Webcam webcam -= Webcam.getDefault()[/code]
should be ...
[code]Webcam webcam = Webcam.getDefault()[/code]

sarxos, 2021-03-04:
@djovla please let us know if the fix proposed by @shinobisoft resolved your issue.

clankill3r, 2021-03-15:
I have the same thing. It should detect 3 webcams but no webcam is detected.
I don't have the ` -= `.

OSX 10.15.7

jpeck22, 2021-03-22:
Same issue.  No cams found.

System.out.println("Webcams: "+Webcam.getDiscoveryService().getWebcams(10, TimeUnit.SECONDS).size() );
     
Webcams: 0

Heriberto-Juarez, 2021-09-29:
Same issue but Im also getting an error regarding SLF4J (Maybe because Im using a Mac with M1 Chip)
Mac OS Big Sur 11.6 

        Webcam webcam = Webcam.getDefault();

        System.out.print("Webcam is ---> ");
        System.out.println(webcam);

SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Webcam is ---> null

I guess I'll try using my Windows machine to check if it works there. (Good luck my app has to work in windows and im not building something for MAC)

eduramiba, 2022-12-09:
Hi, please try this driver, it should work with recent MacOS versions https://github.com/eduramiba/webcam-capture-driver-native