sarxos/webcam-capture #809: unplugging and replugging webcam breaks webcam

TamjidTayeb, 2021-03-04:
Hello,

If i use this library to have take input from a usb webcam everything works fine. But when i unplug the webcam and replug it I am stuck with this message:

[image]

And in the logs I see this:
[image]

How can I stop the webcam from being disposed after being unplugged? Because I think after I replug my webcam it should reconnect to the webcam java input

sarxos, 2021-03-04:
Hello there @TamjidTayeb.

Please take a look at [The Webcam Capture API Handbook](https://github.com/sarxos/webcam-capture/blob/master/webcam-capture/src/etc/resources/webcam-capture-handbook.doc) section 6.1 where _Webcam Discovery Service_ is described.

It explains how to access it, stop, and start it again.

By stopping discovery service your webcam won't be disposed, **but** there is no guarantee if it will be re-instantiated again after it's re-connected. This is completely random and works differently on a different operating systems. I do remember cases where application crashed when I tried to do thinks on a webcam which was disconnected. So, yeah, I'm sure you will have completely new field of scenarios to research and test to identify the problematic ones :)

From my point of view I would solve this by reacting to the disconnection event, e.g. close the panel, and open it again or re-create it again with a new webcam instance after the webcam is connected again.

Here is the sample code to listen for connection/disconnection events (from the handbook).

```java
List<Webcam> webcams = Webcam.getWebcams();
for (Webcam webcam : webcams) {
    // print all currently connected webcams
    System.out.println(webcam.getName());
}

// will print message whenever webcam is connected or disconnected
// the Webcam here is a class, not a webcam instance (addDiscoveryListener is a static method)
Webcam.addDiscoveryListener(
    new WebcamDiscoveryListener() {
        public void webcamFound(WebcamDiscoveryEvent event) {
            System.out.println(event.getWebcam().getName() + " connected");
        }
        public void webcamGone(WebcamDiscoveryEvent event) {
            System.out.println(event.getWebcam().getName() + " disconnected");
        }
    }
);
```

TamjidTayeb, 2021-03-05:
Hi,

Thanks for your quick reply. I have tried to use the webcamdiscoverylistener to cleanly reconnect but i'm still getting issues, could you please have a look at this and let me know what I'm doing wrong?
https://stackoverflow.com/questions/66485557/how-to-handle-webcam-unplug-then-replug-for-sarxos-java-webcam

sarxos, 2021-03-08:
@TamjidTayeb By looking at your code I suppose this happened because you did not remove old `WebcamPanel` from `JFrame` and therefore it displays the old data. You created new one in `webcamFound(..)` but this is also wrong, because you only created it and you did not add it to the `JFrame`. If component is not added to the `JFrame` it will not be displayed (you won't see it).

```java
    @Override
    public void webcamFound( WebcamDiscoveryEvent event ) {
        if( webcam == null ) return;
        Webcam newWebcam = event.getWebcam();
        if( newWebcam.getName().equals( webcam.getName()) ) {
            close();
            webcam = Webcam.getWebcamByName( webcam.getName() );
            webcam.open();
            panel = new WebcamPanel( webcam ); // <--- you need to remove old from the frame and add new
        }
    }
```

You should do similar thing as you did with original `WebcamPanel`:

```java
add( panel );
```

sarxos, 2021-06-12:
I'm closing this ticket. Feel free to reopen it in case of additional questions in regards to the discussed matter.

closed 2021-06-12 by sarxos