sarxos/webcam-capture #810: use ffmpeg driver to capture screen as input

zoff99, 2021-03-08:
awesome library!
can the ffmpeg driver be used to capture the screen at some fps as input source?

shinobisoft, 2021-03-13:
As I'm not the author of this library, I cannot say for sure but I believe that this library isn't capable of doing what you're asking.

zoff99, 2021-03-13:
i already implemented it. but obviously only as PoC

johnpawl, 2021-04-22:
> As I'm not the author of this library, I cannot say for sure but I believe that this library isn't capable of doing what you're asking.

as @shinobisoft said this library isn't capable of doing what you're asking

zoff99, 2021-04-22:
it does , i just quickly hacked it together:

https://github.com/zoff99/java_toxclient_example/tree/master/trifa_desktop/app/src/main/java/com/github/sarxos/webcam/ds/ffmpegcli

alexmao86, 2021-05-10:
Yes, it is supported via invoking native ffmpeg binary as sub process then communicate it in pipeline. see ffmpegcli driver

closed 2021-05-31 by alexmao86