sarxos/webcam-capture #818: can't get more  camer device

hqwl159, 2021-05-06:
When the device number is greater than 15, the device information cannot be obtained.
I can read the camera info on the path of "/dev"
`ls /dev/ |grep video`
```shell
video15
video16
video17
video18
video19
video20
video21
video22
video23
video24
video25
video26
video27
video28
video29
video30
video31
video32
video33
video34
```
I use the webcam-capture code 
```java
       List<Webcam> webcams = Webcam.getWebcams();
     
        System.out.println(webcams.size());

```

The output  is "1".

But I can read  all the  camera device info when the device name ranges from "video0" to "video15".

So I guess the code cannot get device information after “video15”.

Is my code error ?

flx5, 2021-05-07:
That is an issue with the default driver OpenIMAJ. It only scans the device files 0 to 15.
You might want to try the JavaCV driver. In that implementation every device in /dev/video* should be recognized.

hqwl159, 2021-05-07:
> That is an issue with the default driver OpenIMAJ. It only scans the device files 0 to 15. [See here](https://github.com/openimaj/openimaj/blob/df945aea4d037dbfae89ebb475966d1062c1cb63/hardware/core-video-capture/src-native/linux/OpenIMAJGrabber.cpp#L133).
> You might want to try the JavaCV driver. In that implementation every device in /dev/video* should be recognized.

Thank you most sincerely!!!  
The webcam-capture can work without camera driver.  Does the webcam-capture or Linux OS  use the  OpenIMAJ as default driver?

flx5, 2021-05-07:
If you don't specify which driver should be used with setDriver, webcam-capture uses the default builtin driver which is OpenIMAJ regardless of the operation system.

hqwl159, 2021-05-07:
> If you don't specify which driver should be used with setDriver, webcam-capture uses the default builtin driver which is OpenIMAJ regardless of the operation system.

Thanks a lot!!!
I find  it in README.