sarxos/webcam-capture #821: I'm having issues with some libs 😢

LuisHenriqueFA14, 2021-05-26:
I don't use java a lot, so i don't know almost nothing about java (what that i need to compile or what is a .class file), but i'm trying to make a camera app to me, but i can't use it...

I was having some troubles with package, i solved it, and when i was done, it was an error like:

./com/github/sarxos/webcam/Webcam.java:20: error: package org.slf4j does not exist
import org.slf4j.Logger;

So i started my search to solve it. I found a lot of issues about that, and i found something about put the `libs` dir into the classpath (idk where are the classpath lol)

So, i need some help with the libs, like, where to put it ?
I need to put the `org` dir to my root path ?
I need to put the `libs` dir to my root path or the `./com/github/sarxos/webcam/libs`  ?
(Please, be so specific bc i'm dumb)
Thank you, and sorry about my english 🇧🇷

jrbgarcia29, 2021-05-28:
you need to build your project by running "mvn install" in your project base dir so that dependencies will be downloaded and missing jars will be available to your local repository

LuisHenriqueFA14, 2021-05-28:
When i run `mvn install` it returns an error talking `command not found: mvn`.
How can i install it ?
It has another way to build my project ?
Thank you again!

alexmao86, 2021-05-31:
This is not issue, please find java maven guileline by yourself

closed 2021-05-31 by alexmao86