sarxos/webcam-capture #827: Not an Issue: Need Information

mayurb90, 2021-06-07:
Hi,
I am looking for a way to get HardwareId for the webcam using Saroxs library. I can get the camera name but looking for VIDPID of it.
 Any help would be much appreciated . 

Thanks

sarxos, 2021-06-12:
Hi @mayurb90, 

Sorry to disappoint you. Unfortunately, this is not possible. The Webcam Capture API get the webcam name and the webcam ID (nothing fancy, 0 for the 1st device, 1 for the 2nd device, 2 for the 3rd device, etc) from the operating system. These are the only parameters the underlying driver can access from Java. On Linux there is no ID at all, only the camera name, and not always. Sometimes it's only the device file from the `/dev` directory.

closed 2021-06-12 by sarxos

mayurb90, 2021-06-12:
Hi @sarxos, 

Thanks for the information but do you have any suggestion on what could be done in the scenario where system is connected with multiple webcams but we just want to use one for the feed. Also, when you said webcam name from the OS, do we know how OS gets this information, does it have something to do with driver? 

Thanks
Mayur