sarxos/webcam-capture #828: Exception in thread "main" java.lang.NoClassDefFoundError

Maruti-Nandan, 2021-06-12:
I tried to run this code 
```
import com.github.sarxos.webcam.Webcam;
import com.github.sarxos.webcam.WebcamResolution;

import java.io.File;
import java.io.IOException;
import javax.imageio.ImageIO;

public class cam {
    public static void main(String args[]) throws IOException {
        Webcam webc =Webcam.getDefault();  // Get default camera
        if (webc == null) // Can't find camera
            return;

        webc.close();
        webc.setViewSize(WebcamResolution.VGA.getSize());
        webc.open(); // Open the camera
        try { // save the picture captured
            ImageIO.write(webc.getImage(), "PNG", new File("D:/hello.png"));
        } catch (IOException ioe) {
            ioe.printStackTrace();
        }
        webc.close(); // Close the camera
    }    
}
```

sarxos, 2021-06-12:
The SLF4J logging framework JAR is missing in your classpath (with other required dependencies as well, e.g. BridJ). See [how to solve missing LoggerFactory class](https://github.com/sarxos/webcam-capture/wiki/How-to-solve-java.lang.ClassNotFoundException:-org.slf4j.LoggerFactory) wiki page for more details on how you can fix it. Please also check the [download](https://github.com/sarxos/webcam-capture#download) section on the main page for the ZIP. In short - you need to have Webcam Capture JAr but also SLF4J and BridJ (or JARs for other capture framework if you are using something non-default).

closed 2021-06-12 by sarxos

Maruti-Nandan, 2021-06-12:
Thanks for replying

Maruti-Nandan, 2021-06-12:
I added that but now the picture that is captured is totally black.
So, How should I fix that?