sarxos/webcam-capture #830: How to get a specific USB camera?

shichiye, 2021-06-15:
At present, I know that the camera can be obtained by a given name, but I cannot confirm which camera it is. I want to know whether the corresponding USB camera can be obtained by the USB camera similar to the hardware id. I don’t seem to find the corresponding api.

sarxos, 2021-06-15:
@shichiye 

There is no such API in the Webcam Capture project. What we have here is a very basic abstraction over the UVC device. There is no API to get the USB information. Even if we go into the lower level and dive deep into the native C++ code, there is no such thing as well. The operating system abstracts the device and completely decouples it from the USB transport.

One could in theory use some 3rd party library like [usb4java](http://usb4java.org/) to scan the ports and correlate UVC data from Webcam Capture with the hardware device class present in the USB port, but I never tried this and cannot even tell if such thing is possible. This is just a brief idea that came into my mind.

eduramiba, 2021-06-16:
I'm working on publishing as open source a native driver capable of providing actual device ids and very good performance. For the moment it will work on Windows only. Maybe tomorrow it's ready, I will reply back.

We use the driver in a production app and works really well.

eduramiba, 2021-06-17:
Hi,

I just published the driver here: https://github.com/eduramiba/webcam-capture-driver-native

Please let me know if you try it successfully or you have any problem.

To get the device id you will need to cast `WebcamDevice` to `CaptureManagerVideoDevice`

shichiye, 2021-06-18:
@sarxos @eduramiba Thank you both for your answers.
I will try it, and then I will reply to you @eduramiba

pacioc193, 2021-12-10:
Could be possible to know the id device of the USB with the relative istance ?

USB\VID_0C45&PID_6366&MI_00\6&183AF011&0&0000

eduramiba, 2021-12-10:
> Could be possible to know the id device of the USB with the relative istance ?
> 
> USB\VID_0C45&PID_6366&MI_00\6&183AF011&0&0000

Yes, you can with https://github.com/eduramiba/webcam-capture-driver-native and casting `WebcamDevice` to `WebcamDeviceWithId`

pacioc193, 2021-12-10:
Could you help me with an example in pure Java... I tried to follow the tutorial but after loading the dll no output of which webcam is connected to the computer. Without loading dll everything is working well. i tried to contact you on twitter also.

[image]
[image]

pacioc193, 2021-12-10:
Also dll are loaded in the project Netbeans as in photo.
[image]
I also tried to put in same place where opencv read it's own dll. OpenCv could work well without any issue but this driver won't work....

eduramiba, 2021-12-10:
@pacioc193 This library tries to load its DLLs from the natives folder relative to where the program is executed. Adding to netbeans or as jar won't do anything.

Please make sure to run your program in a folder where natives is located. Also I recommend setting up a slf4j backend such as logback for viewing logging messages.

pacioc193, 2021-12-10:
It works. Was a matter of libraries... I'm not using maven project... Standalone application developing here.
Thanks for the support

[image]

eduramiba, 2021-12-10:
@pacioc193 Great! 😄

pacioc193, 2021-12-10:
I think the best ways to solve this issue in the future... release a zip file with all the dependency ahah

closed 2022-05-19 by shichiye