sarxos/webcam-capture #831: UnsatisfiedLinkError

carlos-sa, 2021-06-27:
Hi @sarxos ,

First of all thank you for the wonderful library!

I'm using to develop a simple windows application using desktop compose from jetbrains.

In the development environment the lib is working perfectly, but when I build it for distribution (MSI) any Webcam call throws me an exception.
Sample code:
`       fun getWebcamList(): List<Webcam>? {
            return Webcam.getWebcams()
        }
`

And the exception's stacktrace
`
com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:866)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:845)
	at WebcamManager$Companion.getWebcamList(WebcamManager.kt:9)
	at ui.ComponentsKt.WebcamsDropDown(components.kt:34)
	at ui.ComponentsKt.WebcamInput(components.kt:25)
	at MainKt.mainTab(main.kt:105)
	at MainKt$main$1$2.invoke(main.kt:37)
	at MainKt$main$1$2.invoke(main.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.runtime.AmbientKt.Providers(Ambient.kt:187)
	at androidx.compose.material.TextKt.ProvideTextStyle(Text.kt:246)
	at androidx.compose.material.MaterialThemeKt$MaterialTheme$1.invoke(MaterialTheme.kt:77)
	at androidx.compose.material.MaterialThemeKt$MaterialTheme$1.invoke(MaterialTheme.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.runtime.AmbientKt.Providers(Ambient.kt:187)
	at androidx.compose.material.MaterialThemeKt.MaterialTheme(MaterialTheme.kt:69)
	at MainKt$main$1.invoke(main.kt:36)
	at MainKt$main$1.invoke(main.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.desktop.AppWindowKt$Window$1$1.invoke(AppWindow.kt:83)
	at androidx.compose.desktop.AppWindowKt$Window$1$1.invoke(AppWindow.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.desktop.AppWindow$show$1.invoke(AppWindow.kt:347)
	at androidx.compose.desktop.AppWindow$show$1.invoke(AppWindow.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.runtime.AmbientKt.Providers(Ambient.kt:187)
	at androidx.compose.desktop.AppWindow$onCreate$1.invoke(AppWindow.kt:326)
	at androidx.compose.desktop.AppWindow$onCreate$1.invoke(AppWindow.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.ui.platform.DesktopSelectionKt$DesktopSelectionContainer$4$1.invoke(DesktopSelection.kt:123)
	at androidx.compose.ui.platform.DesktopSelectionKt$DesktopSelectionContainer$4$1.invoke(DesktopSelection.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.ui.layout.LayoutKt.Layout(Layout.kt:682)
	at androidx.compose.ui.platform.DesktopSelectionKt.Wrap(DesktopSelection.kt:38)
	at androidx.compose.ui.platform.DesktopSelectionKt.access$Wrap(DesktopSelection.kt)
	at androidx.compose.ui.platform.DesktopSelectionKt$DesktopSelectionContainer$4.invoke(DesktopSelection.kt:122)
	at androidx.compose.ui.platform.DesktopSelectionKt$DesktopSelectionContainer$4.invoke(DesktopSelection.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt)
	at androidx.compose.runtime.AmbientKt.Providers(Ambient.kt:187)
	at androidx.compose.ui.platform.DesktopSelectionKt.DesktopSelectionContainer(DesktopSelection.kt:121)
	at androidx.compose.ui.platform.DesktopSelectionKt.DesktopSelectionContainer(DesktopSelection.kt:62)
	at androidx.compose.ui.platform.WrapperKt$setContent$1$1.invoke(Wrapper.kt:36)
	at androidx.compose.ui.platform.WrapperKt$setContent$1$1.invoke(Wrapper.kt)
	at androidx.compose.runtime.internal.ComposableLambda.invoke(ComposableLambda.kt:146)
[... 41 lines cut ...]
	... 90 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:45)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:117)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.base/java.lang.Thread.run(Unknown Source)
Caused by: java.lang.UnsatisfiedLinkError: 'int org.bridj.Platform.sizeOf_ptrdiff_t()'
	at org.bridj.Platform.sizeOf_ptrdiff_t(Native Method)
	at org.bridj.Platform.<clinit>(Platform.java:232)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more

`

Any suggestions on this problem?
Thank you!

carlos-sa, 2021-06-27:
Hi!

I was able to fix this issue by updating the compose plugin to version 0.5.0-build225
Don't really know why though

closing the issue for now

thank you

closed 2021-06-27 by carlos-sa

codaman, 2023-08-30:
@carlos-sa did you try with compose on macOs