sarxos/webcam-capture #835: ClassNotFoundException: OpenIMAJGrabber after compiling into jar

MegaBloopy, 2021-08-18:
I am trying to fuse this library with another, and I got it to work, works fine in my IDE, once I compile it into an external jar however, it ends up throwing error after error, I have included the libraries needed for it to function, including the dependencies for the webcam library. But now it gives me a stack trace of `com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124) ~[?:?]
        at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893) ~[?:?]
        at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:956) ~[?:?]
        at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:933) ~[?:?]
        at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:911) ~[?:?]
        at webcam.Main.<init>(Main.java:22) ~[?:?]
        at com.megiryuu.testing.Main.onEnable(Main.java:20) ~[?:?]
        at org.bukkit.plugin.java.JavaPlugin.setEnabled(JavaPlugin.java:263) ~[patched_1.16.5.jar:git-Paper-786]
        at org.bukkit.plugin.java.JavaPluginLoader.enablePlugin(JavaPluginLoader.java:370) ~[patched_1.16.5.jar:git-Paper-786]
        at org.bukkit.plugin.SimplePluginManager.enablePlugin(SimplePluginManager.java:500) ~[patched_1.16.5.jar:git-Paper-786]
        at org.bukkit.craftbukkit.v1_16_R3.CraftServer.enablePlugin(CraftServer.java:518) ~[patched_1.16.5.jar:git-Paper-786]
        at org.bukkit.craftbukkit.v1_16_R3.CraftServer.enablePlugins(CraftServer.java:432) ~[patched_1.16.5.jar:git-Paper-786]
        at net.minecraft.server.v1_16_R3.MinecraftServer.loadWorld(MinecraftServer.java:594) ~[patched_1.16.5.jar:git-Paper-786]
        at net.minecraft.server.v1_16_R3.DedicatedServer.init(DedicatedServer.java:298) ~[patched_1.16.5.jar:git-Paper-786]
        at net.minecraft.server.v1_16_R3.MinecraftServer.w(MinecraftServer.java:1069) ~[patched_1.16.5.jar:git-Paper-786]
        at net.minecraft.server.v1_16_R3.MinecraftServer.lambda$a$0(MinecraftServer.java:291) ~[patched_1.16.5.jar:git-Paper-786]
        at java.lang.Thread.run(Unknown Source) [?:1.8.0_301]
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at java.util.concurrent.FutureTask.report(Unknown Source) ~[?:1.8.0_301]
        at java.util.concurrent.FutureTask.get(Unknown Source) ~[?:1.8.0_301]
        at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116) ~[?:?]
        ... 16 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72) ~[?:?]
        at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140) ~[?:?]
        at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46) ~[?:?]
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:45) ~[?:?]
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:117) ~[?:?]
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36) ~[?:?]
        at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26) ~[?:?]
        at java.util.concurrent.FutureTask.run(Unknown Source) ~[?:1.8.0_301]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source) ~[?:1.8.0_301]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source) ~[?:1.8.0_301]
        ... 1 more
Caused by: java.lang.ExceptionInInitializerError
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55) ~[?:?]
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source) ~[?:1.8.0_301]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source) ~[?:1.8.0_301]
        ... 1 more
Caused by: java.lang.RuntimeException: Failed to register class com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber
        at org.bridj.BridJ.register(BridJ.java:194) ~[?:?]
        at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:60) ~[?:?]
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55) ~[?:?]
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source) ~[?:1.8.0_301]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source) ~[?:1.8.0_301]
        ... 1 more
Caused by: java.lang.ClassNotFoundException: com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber
        at java.net.URLClassLoader.findClass(Unknown Source) ~[?:1.8.0_301]
        at java.lang.ClassLoader.loadClass(Unknown Source) ~[?:1.8.0_301]
        at sun.misc.Launcher$AppClassLoader.loadClass(Unknown Source) ~[?:1.8.0_301]
        at java.lang.ClassLoader.loadClass(Unknown Source) ~[?:1.8.0_301]
        at java.lang.Class.forName0(Native Method) ~[?:1.8.0_301]
        at java.lang.Class.forName(Unknown Source) ~[?:1.8.0_301]
        at org.bridj.BridJ.register(BridJ.java:191) ~[?:?]
        at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:60) ~[?:?]
        at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55) ~[?:?]
        at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81) ~[?:?]
        at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source) ~[?:1.8.0_301]
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source) ~[?:1.8.0_301]`

I am running x64/x86 architecture on Windows 10, running using Java 1.8 v.301

alexmao86, 2021-09-30:
Hi @MegaBloopy ,

Caused by: java.lang.RuntimeException: Failed to register class com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber
        at org.bridj.BridJ.register(BridJ.java:194) ~[?:?]

Please check if native binary, i.e. dll or so lib are packed or is your OS supported.

java.lang.ClassNotFoundException is caused by here:
public class OpenIMAJGrabber extends CPPObject {

	static {
		Platform.addEmbeddedLibraryResourceRoot("com/github/sarxos/webcam/ds/buildin/lib/");
		BridJ.register();
	}
}

rather than OpenIMAJGrabber  not packed.

Alex

xperiosa, 2021-09-30:
Hello i have a similar problem. 
It works fine in a normal maven project(application) but somehow it doesn't work when i try to load it as a jar plugin with pf4j plugin framework(all dependencies are in the jar) Running on linux x64.

       'Webcam webcam = Webcam.getDefault();
        webcam.open();

        BufferedImage image = webcam.getImage();

        try 
        {
            ImageIO.write(image, ImageUtils.FORMAT_JPG, new File("./target/test.jpg"));
        } 
        catch (IOException ex) 
        {
            Logger.getLogger(WebcamPlugin.class.getName()).log(Level.SEVERE, null, ex);
        }
    }`

`30-09-2021 22:53:33 [main] INFO  org.pf4j.AbstractPluginManager - Start plugin 'Webcam-plugin@0.0.1'
30-09-2021 22:53:33 [main] DEBUG org.pf4j.DefaultPluginFactory - Create instance for plugin 'com.webcam.plugin.WebcamPlugin'
SLF4J: Class path contains multiple SLF4J bindings.
SLF4J: Found binding in [jar:file:/home/t7emon/Documents/maven-projects/plugins/Webcam-Plugin-1.0-plugin.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: Found binding in [jar:file:/home/.m2/repository/ch/qos/logback/logback-classic/1.2.3/logback-classic-1.2.3.jar!/org/slf4j/impl/StaticLoggerBinder.class]
SLF4J: See http://www.slf4j.org/codes.html#multiple_bindings for an explanation.
SLF4J: Actual binding is of type [ch.qos.logback.classic.util.ContextSelectorStaticBinder]
30-09-2021 22:53:33 [main] INFO  com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
30-09-2021 22:53:33 [webcam-discovery-service] DEBUG c.g.s.w.d.b.WebcamDefaultDriver - Searching devices
java.util.NoSuchElementException: No ValueType for class org.bridj.Pointer
	at org.bridj.MethodCallInfo.getValueType(MethodCallInfo.java:433)
	at org.bridj.MethodCallInfo.init(MethodCallInfo.java:204)
	at org.bridj.MethodCallInfo.<init>(MethodCallInfo.java:129)
	at org.bridj.MethodCallInfo.<init>(MethodCallInfo.java:90)
	at org.bridj.CRuntime$MethodCallInfoBuilder.apply(CRuntime.java:351)
	at org.bridj.CRuntime.registerNativeMethod(CRuntime.java:504)
	at org.bridj.CRuntime.register(CRuntime.java:454)
	at org.bridj.CRuntime.register(CRuntime.java:345)
	at org.bridj.BridJ.register(BridJ.java:355)
	at org.bridj.BridJ.register(BridJ.java:192)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:60)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
java.util.NoSuchElementException: No ValueType for class org.bridj.Pointer
	at org.bridj.MethodCallInfo.getValueType(MethodCallInfo.java:433)
	at org.bridj.MethodCallInfo.init(MethodCallInfo.java:192)
	at org.bridj.MethodCallInfo.<init>(MethodCallInfo.java:129)
	at org.bridj.MethodCallInfo.<init>(MethodCallInfo.java:90)
	at org.bridj.CRuntime$MethodCallInfoBuilder.apply(CRuntime.java:351)
	at org.bridj.CRuntime.registerNativeMethod(CRuntime.java:504)
	at org.bridj.CRuntime.register(CRuntime.java:454)
	at org.bridj.CRuntime.register(CRuntime.java:345)
	at org.bridj.BridJ.register(BridJ.java:355)
	at org.bridj.BridJ.register(BridJ.java:192)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:60)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
java.util.NoSuchElementException: No ValueType for class org.bridj.Pointer
	at org.bridj.MethodCallInfo.getValueType(MethodCallInfo.java:433)
	at org.bridj.MethodCallInfo.init(MethodCallInfo.java:204)
	at org.bridj.MethodCallInfo.<init>(MethodCallInfo.java:129)
	at org.bridj.MethodCallInfo.<init>(MethodCallInfo.java:90)
	at org.bridj.CRuntime$MethodCallInfoBuilder.apply(CRuntime.java:351)
	at org.bridj.CRuntime.registerNativeMethod(CRuntime.java:504)
	at org.bridj.CRuntime.register(CRuntime.java:454)
	at org.bridj.CRuntime.register(CRuntime.java:345)
[... 78 lines cut ...]
	at org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:64)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more
Caused by: java.lang.RuntimeException: Failed to cast pointer to native object of type org.bridj.cpp.CPPRuntime$CPPDestructor
	at org.bridj.BridJ.createNativeObjectFromPointer(BridJ.java:265)
	at org.bridj.BridJ.createNativeObjectFromPointer(BridJ.java:277)
	at org.bridj.Pointer.getNativeObjectAtOffset(Pointer.java:1900)
	at org.bridj.CommonPointerIOs$CallbackPointerIO.get(CommonPointerIOs.java:191)
	at org.bridj.CommonPointerIOs$CallbackPointerIO.get(CommonPointerIOs.java:178)
	at org.bridj.Pointer.get(Pointer.java:2041)
	at org.bridj.Pointer.get(Pointer.java:2010)
[... cut ...]
Caused by: java.lang.RuntimeException: Failed to cast pointer Pointer(peer = 0x7f63932c2ea0, targetType = org.bridj.cpp.CPPRuntime$CPPDestructor, order = LITTLE_ENDIAN) to instance of type org.bridj.cpp.CPPRuntime$CPPDestructor
	at org.bridj.CRuntime$CTypeInfo.cast(CRuntime.java:209)
	at org.bridj.BridJ.createNativeObjectFromPointer(BridJ.java:259)
	... 22 more
Caused by: java.lang.ClassCastException: org.bridj.cpp.CPPRuntime_CPPDestructor_NativeImpl cannot be cast to org.bridj.NativeObject
	at org.bridj.CRuntime$CTypeInfo.newCastInstance(CRuntime.java:193)
	at org.bridj.CRuntime$CTypeInfo.cast(CRuntime.java:204)
	... 23 more`