sarxos/webcam-capture #838: Connection to IP Cam with Onvif standard with password protection unsuccessful

cinlung, 2021-09-15:
Dear Sarxos

First, I would love to say that your API are amazing for webcam and IP cam. Very easy to learn and easy implementation. With this message, I want to ask your guidance to connecting to the IP camera that I have. It has OnFiv standard and it has web interface protected with username and password. When entering the camera requested me to install VLC, I guess it uses VLC protocol for the image.

Using your example and tutorial for IP Cam connection I was able to connect to a sample camera you provided (http://195.31.81.138/mjpg/video.mjpg), but upon testing with my camera it is stuck with no error whatsoever.

Here are parts of my codes:
`System.out.println("Adding driver for ip cam...");
    Webcam.setDriver(new IpCamDriver());
    
    System.out.println("Registering a camera...");
    //IpCamDeviceRegistry.register(new IpCamDevice("SafePark", "http://195.31.81.138/mjpg/video.mjpg", IpCamMode.PUSH));
    IpCamDeviceRegistry.register(new IpCamDevice(devName, url, mode));
    
    System.out.println("Preparing the output frame...");
    JFrame f = new JFrame(title);
    f.setLayout(new FlowLayout());
    
    System.out.println("Adding webcam panel view...");
    Webcam webcam = Webcam.getDefault();
    System.out.println("Adding webcam panel view... 2");
    WebcamPanel panel = new WebcamPanel(webcam);
    System.out.println("Adding webcam panel view... 3");
    f.add(panel);`

The code stuck at WebcamPanel panel = new WebcamPanel(webcam);

Would you please tell me what is wrong with my camera? Is there additional steps needed? Thank you for your helps in advance.

alexmao86, 2021-09-30:
it happens here：
`	public WebcamPanel(Webcam webcam, Dimension size, boolean start, ImageSupplier supplier) {

		if (webcam == null) {
			throw new IllegalArgumentException(String.format("Webcam argument in %s constructor cannot be null!", getClass().getSimpleName()));
		}

		this.defaultSize = size;
		this.webcam = webcam;
		this.updater = new ImageUpdater();
		this.supplier = supplier;
		this.rb = WebcamUtils.loadRB(WebcamPanel.class, getLocale());

		setDoubleBuffered(true);

		addPropertyChangeListener("locale", this);

		if (size == null) {
			Dimension r = webcam.getViewSize();
			if (r == null) {
				r = webcam.getViewSizes()[0];
			}
			setPreferredSize(r);
		} else {
			setPreferredSize(size);
		}

		if (start) {
			start();
		}
	}`

alexmao86, 2021-09-30:
it happens:

	public WebcamPanel(Webcam webcam, Dimension size, boolean start, ImageSupplier supplier) {

		if (webcam == null) {
			throw new IllegalArgumentException(String.format("Webcam argument in %s constructor cannot be null!", getClass().getSimpleName()));
		}

		this.defaultSize = size;
		this.webcam = webcam;
		this.updater = new ImageUpdater();
		this.supplier = supplier;
		this.rb = WebcamUtils.loadRB(WebcamPanel.class, getLocale());

		setDoubleBuffered(true);

		addPropertyChangeListener("locale", this);

		if (size == null) {
			Dimension r = webcam.getViewSize();
			if (r == null) {
				r = webcam.getViewSizes()[0];
			}
			setPreferredSize(r);
		} else {
			setPreferredSize(size);
		}

		if (start) {
			start(); ----------------------------------------->>>>>start
		}
	}

start will goto IPCamera driver:

	@Override
	public void open() {
		if (!open) {

			reader = createReader();
			reader.start();

			try {
				reader.readImage(); -------------------------------------->>>>>blocking here
			} catch (InterruptedException e) {
				throw new WebcamException(e);
			}
		}
		open = true;
	}

so please check your MJPEG stream server is qualified server.

Alex

cinlung, 2021-09-30:
Hi Alex, 

 

Thanks for the information. I actually able to perform IP Cam Streaming. The solution was to find the appropriate URL for the MJPEG service in the camera. Sorry I was not able to reply back because I forgot the link and was distracted by other stuff. But, I appreciate the hints.

 

Thank you