sarxos/webcam-capture #848: How to call Webcam.dispose() to release it

stefan-reich, 2022-01-25:
Hi,

quick question about your lovely library. I need a way to release the web cam from the current process so another process can access it. (On Linux, only one process can access a webcam at a time.)

The method Webcam.dispose() probably does this. But it's protected, so I don't know how to call it. I could call using reflection but is that necessary?

I also found WebcamDiscoveryService and WebcamDeallocator which indirectly call Webcam.dispose(), but they don't seem suited to what I want to do. I simply want to release a Webcam that I hold a reference to.

Many greetings

MrFirekiller, 2022-01-27:
It's just webcam.close()

stefan-reich, 2022-01-28:
Alright that was easy then :)

closed 2022-01-28 by stefan-reich

MrFirekiller, 2022-01-28:
A good tip is to press the dot before the method name, eg webcam. and then scroll to see all the methods available. It's handy like this. Also see the examples.

stefan-reich, 2022-01-28:
Yeah that's cute, thanks :) I'm a Java developer since 1998. I was just asking if the webcam is actually released to the OS like this. Could have tested I guess. Greetings

MrFirekiller, 2022-01-29:
Oh ok I just misunderstood you I guess

stefan-reich, 2022-01-29:
@MrFirekiller No problem