sarxos/webcam-capture #851: not getting access to Camera in Mac-Latest versions using Webcam.getWebcams(), or Webcam.getDefaultCam()

svcsrao, 2022-02-08:
Tried many ways using saros 0.3.12 jar and not able to access webcam (both internal and external) in latest MAC OS versions.

Erhannis, 2022-02-08:
Same here.  Here's some other similar tickets; seems like Mac has been difficult to maintain.
https://github.com/sarxos/webcam-capture/issues/808
https://github.com/sarxos/webcam-capture/issues/766
https://github.com/sarxos/webcam-capture/issues/757
https://github.com/sarxos/webcam-capture/issues/723

Though actually, it looks like the final post in #757 ( https://github.com/sarxos/webcam-capture/issues/757#issuecomment-849632328 ) works - I don't know which components are strictly necessary, but after swapping out my dependencies for the ones that look relevant, and adding the repositories, my test code compiled and ran, apparently correctly fetching a default webcam (which was where I was stuck before).

Erhannis, 2022-02-08:
I removed the dependency on `webcam-capture-driver-openimaj`, and it still fetched a webcam correctly, but both the repositories seem necessary, and presumably both `webcam-capture` and `bridj` are necessary.

eduramiba, 2022-12-09:
Hi, please try this driver, it should work with recent MacOS versions https://github.com/eduramiba/webcam-capture-driver-native