sarxos/webcam-capture #856: Webcam Not Detecting in macOs BigSur

Robinbr21, 2022-03-12:
in my macOs BigSur version 11.6 webcam was not detecting 
i'am using  webcam-capture-0.3.12.jar
is there any way to use  webcam-capture in mac please suggest (or any other way in java)

**_18986 [Thread-8] INFO com.github.sarxos.webcam.Webcam - WebcamDefaultDriver capture driver will be used
20088 [Thread-8] WARN com.github.sarxos.webcam.Webcam - No webcam has been detected!_**

JoshuaOlu, 2022-03-13:
I'm having this same issue in my macOs BigSur version 11.1. Is there any solution?

Robinbr21, 2022-03-14:
Tried many. No solutions yet! 

@sarxos

JoshuaOlu, 2022-04-08:
Is this going to be solved or not?

anthonydonx, 2022-04-28:
having same issue, no solution yet

Robinbr21, 2022-04-29:
no didn't get any solution .......

GuoLiuFang, 2022-05-17:
@alexflex25 **It works**.
12.2.1

Robinbr21, 2022-05-20:
@alexflex25  i dont understand what u said  .

@GuoLiuFang  which one works??

please mention below

alexflex25, 2022-05-31:
@Robinbr21 This is a gradle configuration file to get a particular version of the jar.
The snapshot version and a specific version of bridJ

webcam-capture-0.3.13-SNAPSHOT.jar
bridj-0.7-20140918-3.jar

GuoLiuFang, 2022-06-11:
> @alexflex25 i dont understand what u said .
> 
> @GuoLiuFang which one works??
> 
> please mention below
this works.
[image]

sbahloul, 2022-06-15:
Thank you for the version published as 0.3.13-SNAPSHOT. Any chance to get a release with this version ?

eduramiba, 2022-12-09:
Hi, please try this driver, it should work with recent MacOS versions https://github.com/eduramiba/webcam-capture-driver-native