sarxos/webcam-capture #859: Webcam not streaming second time

svcsrao, 2022-04-27:
for Mac I am using "OpenImajDriver(); ==> it is streaming perfectly every time.

for Windows I am using "CaptureManagerDriver()" ==> it is streaming first time and from second time onwards, it is not streaming and displaying last captured image.

First time camera opening (Displaying the led light), second time onwards it is not displaying the led light

eduramiba, 2022-12-09:
Can you check if disabling locking works?

`webcam.getLock().disable();`

Also, for a better Mac driver, please try the recent update of https://github.com/eduramiba/webcam-capture-driver-native

alexxRizz, 2025-01-31:
> Can you check if disabling locking works?
> 
> `webcam.getLock().disable();`
> 
> Also, for a better Mac driver, please try the recent update of https://github.com/eduramiba/webcam-capture-driver-native

I have the same problem. If i reconnect my camera locking exception occures. And disabling locking helped and connection reastablished! Thank you.