sarxos/webcam-capture #860: macOS 12: com.github.sarxos.webcam.WebcamException: Cannot execute task

NickJi2019, 2022-05-09:
# I just wrote Webcam.getWebcams() in the main function.

Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:866)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:845)
	at com.woznes.facerec_v6.main.FaceRecApplicationKt.main(FaceRecApplication.kt:24)
	at com.woznes.facerec_v6.main.FaceRecApplicationKt.main(FaceRecApplication.kt)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.base/java.util.concurrent.FutureTask.report(FutureTask.java:122)
	at java.base/java.util.concurrent.FutureTask.get(FutureTask.java:191)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 5 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.getDevices(WebcamDefaultDriver.java:79)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:124)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:26)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1136)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:635)
	at java.base/java.lang.Thread.run(Thread.java:833)
Caused by: java.lang.UnsatisfiedLinkError: 'org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices()'
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices(Native Method)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$GetDevicesTask.handle(WebcamDefaultDriver.java:93)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	... 3 more

alexflex25, 2022-05-10:
Did you try this:
`Webcam.getDiscoveryService().setEnabled(false);`

0xD8C4A475, 2022-08-03:
> 

same.

Problem is arm64 but loaded linux_x64

Half solution: https://github.com/languagetool-org/languagetool/issues/4543#issuecomment-829037244

But next issue: https://github.com/languagetool-org/languagetool/issues/4543#issuecomment-1203919691

ywyang236, 2023-05-16:
> Did you try this: `Webcam.getDiscoveryService().setEnabled(false);`

@alexflex25 I am a beginner. Would you mind provide a more detailed description? Thank you for your help.