sarxos/webcam-capture #862: Webcam Library

sisco12, 2022-06-08:
II am trying this library and after adding all the jar files. I have tried this easy code :

```
public static void main(String[] args) {
		Webcam webcam = Webcam.getDefault();
		if (webcam != null) {
			System.out.println("Webcam: " + webcam.getName());
		} else {
			System.out.println("No webcam detected");
		}
	}
```
 
I get this error :
```
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:956)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:933)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:911)
```

sisco12, 2022-06-13:
> Can you give me more information? In particular what OS you are on? what version of Java etc...

Using MacOS Montery version 12.0.1
java -version :
openjdk version "1.8.0_302"
OpenJDK Runtime Environment (Temurin)(build 1.8.0_302-b08)
OpenJDK 64-Bit Server VM (Temurin)(build 25.302-b08, mixed mode)

alexflex25, 2022-06-15:
Did you try this version?

https://github.com/sarxos/webcam-capture/issues/856#issuecomment-1122443353