sarxos/webcam-capture #870: is it support aarch64

kuaikuaikim, 2022-08-04:
is it support aarch64? which driver should i select in aarch64.

0xD8C4A475, 2022-08-08:
Look:
https://github.com/sarxos/webcam-capture/tree/master/webcam-capture/src/main/resources/com/github/sarxos/webcam/ds/buildin/lib

alexmao86, 2022-08-10:
hi Kuaikuaikim,
The buit-in dirver does not support aarch64  caused by limitation of bridj and imag native resources. You can try opencv driver, not much effort can run up aarch64.
thanks,
Alex

Jeizz, 2022-11-17:
> hola Kuaikuaikim, El controlador incorporado no es compatible con aarch64 debido a la limitación de los recursos nativos de bridj e imag. Puede probar el controlador opencv, no se puede hacer mucho esfuerzo con aarch64. gracias, alex

is it possible to add what is necessary to make OpenImaj work on aarch64 ?

recently bridj includes the files for aarch64 in its latest version

kuaikuaikim, 2022-11-17:
这是来自QQ邮箱的假期自动回复邮件。你好，我最近正在休假中，无法亲自回复你的邮件。我将在假期结束后，尽快给你回复。

alexmao86, 2023-12-05:
HI ,
the JNI part is relay on BridJ, but BridJ stopped for a long time, if you want to support aarch64 , you can cross compile native so lib of use other camera driver, the opencv one support it.

closed 2023-12-05 by alexmao86