sarxos/webcam-capture #872: Not working on Mac devices

jlTorresAdCentral, 2022-09-12:
Hi I'm working on a multiplatform project with windows and macOS using intellij.

I have no problem on window, it works perfectly, but on mac i'm trying to get the default camera and the app crashes

`val webcam = Webcam.getDefault()`

I'm getting this error
`Caused by: java.lang.UnsatisfiedLinkError: 'org.bridj.Pointer com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.getVideoDevices()'`

macOS Monterey 12.3.1

runnermann, 2022-10-13:
We are having the same issue at FlashMonkey. Windows works perfectly. However, macOS Monterey does not allow access even after user has granted permission. 

We've attempted using version 0.3.12 and 0.3.13-SNAPSHOT.

macOS Monterey 12.6

eduramiba, 2022-12-09:
Hi, please try this driver, it should work with mac https://github.com/eduramiba/webcam-capture-driver-native

msgilligan, 2022-12-10:
>  https://github.com/eduramiba/webcam-capture-driver-native

This driver does work with macOS (but not for all dimensions), but I ran into Issue #889 while trying to get it to work.