sarxos/webcam-capture #876: Webcam discovery Listener

meeeee12, 2022-10-09:
When I pull out a the active webcam's USB cable it takes around 5 seconds to realise the webcam has been removed which is ok, but If I plug the webcam back in before the `webcamGone` discovery event is triggered, the `webcamGone` and `webcamFound` events are never triggered. The WebcamPanel no longer updates and I keep getting `[frames-refresher-[0]] ERROR com.github.sarxos.webcam.ds.buildin.WebcamDefaultDevice - Timeout when requesting image!`. Is it possible to avoid this behaviour?

meeeee12, 2022-10-09:
I am using Java 11.0.9 running on Windows 10

diveair, 2022-11-02:
I too am looking for a solution to this issue, and am hoping to edit the library myself to include a fix but I cannot get a dev environment without dependencies issues which seems to throw the exception:

java.io.FileNotFoundException: Library 'OpenIMAJGrabber' was not found in path '[., C:/Program Files/Java/jre1.8.0_161/bin/server, C:/Program Files/Java/jre1.8.0_161/bin, C:/Program Files/Java/jre1.8.0_161/lib/amd64, C:\ProgramData\Oracle\Java\javapath, C:\Windows\system32, C:\Windows, C:\Windows\System32\Wbem, C:\Windows\System32\WindowsPowerShell\v1.0\, C:\Windows\System32\OpenSSH\, C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL, C:\Program Files\Intel\Intel(R) Management Engine Components\DAL, C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common, C:\WINDOWS\system32, C:\WINDOWS, C:\WINDOWS\System32\Wbem, C:\WINDOWS\System32\WindowsPowerShell\v1.0\, C:\WINDOWS\System32\OpenSSH\, C:\Program Files\Intel\WiFi\bin\, C:\Program Files\Common Files\Intel\WirelessCommon\, C:\Program Files (x86)\Schneider Electric\SoMachine Software\Tools\ConfigurationManager\, C:\Program Files (x86)\QuickTime\QTSystem\, C:\Program Files\Java\jdk1.8.0_161, C:\Program Files\Java\jdk1.8.0_161\bin, C:\Program Files\WorldPainter, C:\Program Files\TortoiseSVN\bin, C:\Program Files\dotnet\, C:\Users\PC\AppData\Local\Microsoft\WindowsApps, C:\Users\PC\Desktop\eclipse, C:\Program Files\Java\jre1.8.0_161\bin, C:\WINDOWS\system32, C:\WINDOWS, C:/Program Files/Java/jre1.8.0_161/bin/server, C:/Program Files/Java/jre1.8.0_161/bin, C:/Program Files/Java/jre1.8.0_161/lib/amd64, C:\ProgramData\Oracle\Java\javapath, C:\Windows\system32, C:\Windows, C:\Windows\System32\Wbem, C:\Windows\System32\WindowsPowerShell\v1.0\, C:\Windows\System32\OpenSSH\, C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL, C:\Program Files\Intel\Intel(R) Management Engine Components\DAL, C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common, C:\WINDOWS\system32, C:\WINDOWS, C:\WINDOWS\System32\Wbem, C:\WINDOWS\System32\WindowsPowerShell\v1.0\, C:\WINDOWS\System32\OpenSSH\, C:\Program Files\Intel\WiFi\bin\, C:\Program Files\Common Files\Intel\WirelessCommon\, C:\Program Files (x86)\Schneider Electric\SoMachine Software\Tools\ConfigurationManager\, C:\Program Files (x86)\QuickTime\QTSystem\, C:\Program Files\Java\jdk1.8.0_161, C:\Program Files\Java\jdk1.8.0_161\bin, C:\Program Files\WorldPainter, C:\Program Files\TortoiseSVN\bin, C:\Program Files\dotnet\, C:\Users\PC\AppData\Local\Microsoft\WindowsApps, C:\Users\PC\Desktop\eclipse, ., C:\Program Files\Java\jre1.8.0_161\bin, C:\Program Files\Java\jre1.8.0_161\bin]'
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1074)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1049)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:602)
	at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:771)
	at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:1022)
	at org.bridj.cpp.CPPRuntime$CPPTypeInfo.initialize(CPPRuntime.java:904)
	at org.bridj.CRuntime$CTypeInfo.initialize(CRuntime.java:271)
	at org.bridj.BridJ.initialize(BridJ.java:1128)
	at org.bridj.NativeObject.<init>(NativeObject.java:50)
	at org.bridj.StructObject.<init>(StructObject.java:46)
	at org.bridj.cpp.CPPObject.<init>(CPPObject.java:55)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<init>(OpenIMAJGrabber.java:62)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Exception in thread "main" com.github.sarxos.webcam.WebcamException: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:124)
	at com.github.sarxos.webcam.Webcam.getWebcams(Webcam.java:893)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:956)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:933)
	at com.github.sarxos.webcam.Webcam.getDefault(Webcam.java:911)
	at guitest.Widgets.WebcamWidget.<init>(WebcamWidget.java:60)
	at guitest.MainApp.main(MainApp.java:75)
Caused by: java.util.concurrent.ExecutionException: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at java.util.concurrent.FutureTask.report(Unknown Source)
	at java.util.concurrent.FutureTask.get(Unknown Source)
	at com.github.sarxos.webcam.WebcamDiscoveryService.getWebcams(WebcamDiscoveryService.java:116)
	... 6 more
Caused by: com.github.sarxos.webcam.WebcamException: Cannot execute task
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.process(WebcamProcessor.java:72)
	at com.github.sarxos.webcam.WebcamProcessor.process(WebcamProcessor.java:140)
	at com.github.sarxos.webcam.WebcamTask.process(WebcamTask.java:46)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.newGrabber(WebcamDefaultDriver.java:45)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver.getDevices(WebcamDefaultDriver.java:117)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:36)
	at com.github.sarxos.webcam.WebcamDiscoveryService$WebcamsDiscovery.call(WebcamDiscoveryService.java:1)
[... cut ...] 161\bin, C:\Program Files\WorldPainter, C:\Program Files\TortoiseSVN\bin, C:\Program Files\dotnet\, C:\Users\PC\AppData\Local\Microsoft\WindowsApps, C:\Users\PC\Desktop\eclipse, ., C:\Program Files\Java\jre1.8.0_161\bin, C:\Program Files\Java\jre1.8.0_161\bin]'
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1074)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:1049)
	at org.bridj.BridJ.getNativeLibrary(BridJ.java:602)
	at org.bridj.cpp.CPPRuntime.newCPPInstance(CPPRuntime.java:771)
	... 13 more