sarxos/webcam-capture #878: maven project call webcam-capture fail on aarch64 platform

dshup, 2022-10-10:
maven project call webcam-capture fail on aarch64 platform
get exceptions:

java.lang.RuntimeException: Failed to initialize BridJ (java.lang.UnsatisfiedLinkError: /tmp/BridJExtractedLibraries2069336107326078457/libbridj.so: /tmp/BridJExtractedLibraries2069336107326078457/libbridj.so: 无法打开共享对象文件: 没有那个文件或目录 (Possible cause: can't load AMD 64-bit .so on a AARCH64-bit platform))
	at org.bridj.Platform.initLibrary(Platform.java:444)
at org.bridj.Platform.<clinit>(Platform.java:228)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)

Caused by: java.lang.UnsatisfiedLinkError: org.bridj.Platform.sizeOf_ptrdiff_t()I
	at org.bridj.Platform.sizeOf_ptrdiff_t(Native Method)
	at org.bridj.Platform.<clinit>(Platform.java:232)
	at com.github.sarxos.webcam.ds.buildin.natives.OpenIMAJGrabber.<clinit>(OpenIMAJGrabber.java:59)
	at com.github.sarxos.webcam.ds.buildin.WebcamDefaultDriver$WebcamNewGrabberTask.handle(WebcamDefaultDriver.java:55)
	at com.github.sarxos.webcam.WebcamProcessor$AtomicProcessor.run(WebcamProcessor.java:81)

my maven config:

`<dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture</artifactId>
            <version>0.3.12</version>
        </dependency>

        <dependency>
            <groupId>com.nativelibs4java</groupId>
            <artifactId>bridj</artifactId>
            <version>0.7.0</version>
        </dependency>`

dshup, 2022-10-10:
maven config:

        <dependency>
            <groupId>com.github.sarxos</groupId>
            <artifactId>webcam-capture</artifactId>
            <version>0.3.12</version>
        </dependency>

        <dependency>
            <groupId>com.nativelibs4java</groupId>
            <artifactId>bridj</artifactId>
            <version>0.7.0</version>
        </dependency>

shinobisoft, 2022-10-10:
See this [https://github.com/sarxos/webcam-capture/issues/870#issuecomment-1210795046]()

alexmao86, 2022-10-23:
offical OpenImaj does not include aarch64, i.e. arm64.

dshup, 2022-10-25:
> offical OpenImaj does not include aarch64, i.e. arm64.

I got it, thank you! I already changed to use opencv